rerun-io/rerun #9090: Add support for fish eye cameras

adhityabg, 2025-02-20:
**Is your feature request related to a problem? Please describe.**
When visualising images, I have to constantly run a un-distortion algorithm to correct the image, so that I can use the Pinhole 3D archetype to view the image in 3D space and to also project the lidar points on the image itself. However, this un-distortion takes time and it slows down the loading of the data in rerun. 

**Describe the solution you'd like**
It will be nice to have a fisheye camera archetype that can display fisheye camera images on to the 3D world and instead apply the same distortion to the lidar points (or any other 3D elements) when projecting on the image.

Wumpf, 2025-02-20:
Definitely something we want to get support for in! 
However, closing it as a duplicate of
* https://github.com/rerun-io/rerun/issues/2315

feel free to comment (and upvote!) the older ticket

closed 2025-02-20 by Wumpf

adhityabg, 2025-02-20:
Ah, sorry

Wumpf, 2025-02-20:
no problem, lotsa tickets easy to miss things