rerun-io/rerun #9159: Rerun library doesn't work with current `arrow-rs` version

manglemix, 2025-02-27:
**Describe the bug**
There was a recent commit that broke `arrow-rs`, which Rerun depends on.
https://github.com/apache/arrow-rs/issues/7209

**To Reproduce**
Steps to reproduce the behavior:
1. `cargo check rerun`

**Backtrace**
```
error[E0034]: multiple applicable items in scope
   --> /home/manglemix/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/arrow-arith-53.4.0/src/temporal.rs:91:36
    |
91  |         DatePart::Quarter => |d| d.quarter() as i32,
    |                                    ^^^^^^^ multiple `quarter` found
    |
note: candidate #1 is defined in the trait `ChronoDateExt`
   --> /home/manglemix/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/arrow-arith-53.4.0/src/temporal.rs:637:5
    |
637 |     fn quarter(&self) -> u32;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^
note: candidate #2 is defined in the trait `Datelike`
   --> /home/manglemix/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/chrono-0.4.40/src/traits.rs:47:5
    |
47  |     fn quarter(&self) -> u32 {
    |     ^^^^^^^^^^^^^^^^^^^^^^^^
help: disambiguate the method for candidate #1
    |
91  |         DatePart::Quarter => |d| ChronoDateExt::quarter(&d) as i32,
    |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
help: disambiguate the method for candidate #2
    |
91  |         DatePart::Quarter => |d| Datelike::quarter(&d) as i32,
    |                                  ~~~~~~~~~~~~~~~~~~~~~
```

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.6.1

**Rerun version**
`
rerun = { version = "0.22.0", features = ["nasm", "sdk"], default-features = false }
`

**Additional context**
I couldn't find any statement addressing this issue

alamb, 2025-02-27:
Note we have released a new version of arrow-rs that fixes hte issue

Wumpf, 2025-03-03:
Can confirm this happens in a minimal application with
```
[dependencies]
chrono = "0.4.40"
rerun = { version = "0.22.0", features = [
    "nasm",
    "sdk",
], default-features = false }
```

@manglemix you can work around this by forcing the `chrono` patch version to an older version. `cargo update chrono --precise 0.4.34` worked for me

@alamb any chance that fix could be backported to `53.4.x`? That's what the `rerun` crate depends on currently :/

alamb, 2025-03-03:
> Can confirm this happens in a minimal application with
>
> ```
> [dependencies]
> chrono = "0.4.40"
> rerun = { version = "0.22.0", features = [
>     "nasm",
>     "sdk",
> ], default-features = false }
> ``` [...]

I would be willing to make such a release.

Could you help by:
1. Make a PR to backport the fix to  https://github.com/apache/arrow-rs/tree/53.0.0_maintenance
2. File a tracking ticket to create a 53.4.1 release (e.g. copy/paste/modify the one from https://github.com/apache/arrow-rs/issues/7209)

emilk, 2025-03-04:
* PR: https://github.com/apache/arrow-rs/pull/7231

edgarriba, 2025-03-23:
hi, this is still happening to use kornia-rs. We had to downgrade everywhere chrono to 0.4.39. Are you planning any fix from rerun to avoid this ? See: https://github.com/kornia/kornia-rs/pull/308

emilk, 2025-03-25:
The issue is resolved on Rerun `main` (I just tested) and so will be fixed in the next Rerun release: https://github.com/rerun-io/rerun/milestone/37

closed 2025-03-25 by emilk