rerun-io/rerun #9162: xkbcommon: ERROR

behnamasadi, 2025-02-27:
Ubuntu 24.04, rerun, installation via Anaconda, 
Running `rerun` crashes with the followings:
```
rerun 
xkbcommon: ERROR: failed to add default include path /home/behnam/anaconda3/envs/OpenCVProjects/share/X11/xkb

thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: XKBNotFound'
```

Solution:
First install 
```
sudo apt install xkb-data libxkbcommon0 libxkbcommon-dev
``` 
and then:

```
export XKB_CONFIG_ROOT=/usr/share/X11/xkb
```

abey79, 2025-03-03:
This seems to be covered by [this documentation page](https://rerun.io/docs/getting-started/troubleshooting#running-on-linux). Is there anything that it's missing? My understanding is that, in general, `XKB_CONFIG_ROOT` doesn't need to be set in regular installation (`/usr/share/X11/xkb`)

behnamasadi, 2025-03-03:
@abey79 Thanks for your reply, I guess the issue is integration with "Anaconda" that forced me to 

```
export XKB_CONFIG_ROOT=/usr/share/X11/xkb
```
I just wrote this issue just in case someone else install it via anaconda, and face the same problem won't get lost, happy rerun!

closed 2025-03-03 by abey79