rerun-io/rerun #9170: add attributes for `class Boxes3D` and `class LineStrips3D`

Qubi0-0, 2025-03-03:
It would be extremely helpful if there was an option to add at least `str` as and attribute to an instance. When you click, it would show on the right extra information. Some parameters, results, values in general. There is already `label` and `class_id` however i feel extra place for more information would be very useful. 

It would definitely make Rerun more interactive.

Wumpf, 2025-03-03:
@Qubi0-0  Not quite following what the feature request here is. If you want to add arbitrary data to an entity, you can always use [`AnyValues`](https://rerun.io/docs/howto/logging/custom-data) which then will show up on hover and on selection of said entity.
Are you looking for something more specific?

Qubi0-0, 2025-03-04:
Amazing! That is exactly what I needed! Thanks!

closed 2025-03-04 by Qubi0-0