rerun-io/rerun #9173: Chunks are duplicated instead of compacted

emilk, 2025-03-03:
## Repro
Step one: generate some test data, e.g. with

`pixi run -e py docs/snippets/compare_snippet_output.py archetypes/scalar_simple`

Or just run this:

```rs
fn main() -> Result<(), Box<dyn std::error::Error>> {
    let rec = rerun::RecordingStreamBuilder::new("rerun_example_scalar").spawn()?;

    // Log the data on a timeline called "step".
    for step in 0..64 {
        rec.set_time_sequence("step", step);
        rec.log("scalar", &rerun::Scalar::new((step as f64 / 10.0).sin()))?;
    }

    Ok(())
}
```

Step two: print it with:

`pr rerun rrd print -vvv docs/snippets/all/archetypes/scalar_simple_rust.rrd`

## Expected
One chunk with 64 rows

## Actual
128 rows (!). Each log call becomes two identical chunks (identical in all but ID).

Notice how `log_tick` increments every _other_ row:

```
StoreInfo {
    application_id: ApplicationId(
        "rerun_example_scalar",
    ),
    store_id: StoreId {
        kind: Recording,
        id: "db153811-a214-45d6-9fce-b4d190a1cf92",
    },
    cloned_from: None,
    is_official_example: false,
    started: "16:15:11.710398Z",
    store_source: RustSdk {
        rustc_version: "1.81.0 (eeb90cda1 2024-09-04)",
        llvm_version: "18.1.7",
    },
    store_version: Some(
        CrateVersion {
            major: 0,
            minor: 23,
            patch: 0,
            meta: Some(
                DevAlpha {
                    alpha: 1,
                    commit: None,
                },
            ),
        },
    ),
}
Store(db153811-a214-45d6-9fce-b4d190a1cf92), Chunk(1829571D292DC534263d66eb1e80213a) with 1 rows (2.3 KiB) - /scalar - 
┌─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
│ METADATA:                                                                                                                                   │
│ * entity_path: "/scalar"                                                                                                                    │
│ * heap_size_bytes: "1160"                                                                                                                   │
│ * id: "1829571D292DC534263d66eb1e80213a"                                                                                                    │
│ * version: "1"                                                                                                                              │
[... 50 lines cut ...]
│ ╞══════════════════════════════════╪════════════════════════╪════════════════════════════╪════════════════════╪═══════════════════════════╡ │
│ │ 1829571D292D185A3461b5a5a9f5d8da ┆ 1                      ┆ 2025-03-03T16:15:13.711899 ┆ 1                  ┆ [0.09983341664682815]     │ │
│ └──────────────────────────────────┴────────────────────────┴────────────────────────────┴────────────────────┴───────────────────────────┘ │
└─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

Store(db153811-a214-45d6-9fce-b4d190a1cf92), Chunk(1829571D2931CC9E263d66eb1e80213d) with 1 rows (1.6 KiB) - /scalar - 
┌────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
│ METADATA:                                                                                                                                      │
│ * entity_path: "/scalar"                                                                                                                       │
│ * heap_size_bytes: "784"                                                                                                                       │
│ * id: "1829571D2931CC9E263d66eb1e80213d"                                                                                                       │
│ * version: "1"                                                                                                                                 │
├╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┤
│ ┌──────────────────────────────────┬────────────────────────┬────────────────────────────┬────────────────────┬──────────────────────────────┐ │
│ │ RowId                            ┆ log_tick               ┆ log_time                   ┆ step               ┆ ScalarIndicator              │ │
│ │ ---                              ┆ ---                    ┆ ---                        ┆ ---                ┆ ---                          │ │
│ │ type: "FixedSizeBinary[16]"      ┆ type: "i64"            ┆ type: "Timestamp(ns)"      ┆ type: "i64"        ┆ type: "List[null]"           │ │
│ │ ARROW:extension:name: "TUID"     ┆ index_name: "log_tick" ┆ index_name: "log_time"     ┆ index_name: "step" ┆ component: "ScalarIndicator" │ │
│ │ is_sorted: "true"                ┆ is_sorted: "true"      ┆ is_sorted: "true"          ┆ is_sorted: "true"  ┆ kind: "data"                 │ │
│ │ kind: "control"                  ┆ kind: "index"          ┆ kind: "index"              ┆ kind: "index"      ┆                              │ │
│ ╞══════════════════════════════════╪════════════════════════╪════════════════════════════╪════════════════════╪══════════════════════════════╡ │
│ │ 1829571D2931CC9E263d66eb1e80213e ┆ 1                      ┆ 2025-03-03T16:15:13.711899 ┆ 1                  ┆ [null]                       │ │
│ └──────────────────────────────────┴────────────────────────┴────────────────────────────┴────────────────────┴──────────────────────────────┘ │
└────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

jleibs, 2025-03-03:
I am only able to reproduce this by setting: `export RERUN_FLUSH_NUM_ROWS=0`

jleibs, 2025-03-03:
Your description of the problem is incorrect -- this is not two identical chunks. Scroll to the end of the column.

For each time point it's:
- One "Scalar" chunk
- One "ScalarIndicator" chunk

emilk, 2025-03-03:
Ah - that makes sense. My bad 🤦

closed 2025-03-03 by emilk