rerun-io/rerun #9185: Web Viewer Fails to Resolve .wasm Path Correctly in Production Builds (Vite/Quasar)

carueda, 2025-03-04:
**Describe the bug**

In Quasar (Vue 3) + Vite, upgrading @rerun-io/web-viewer from 0.16.1 to 0.17.1 (or later, including 0.22.1) causes a malformed URL issue when trying to load the .wasm file.

- The .wasm file exists and is correctly served in production builds.
- However, Rerun constructs an invalid URL for it, leading to a `TypeError: Failed to construct 'URL'` error.
- Rerun seems to use `import.meta.url` incorrectly inside a `new URL()` call, which doesn't work in production builds where assets are optimized by Vite.

**To Reproduce**

1. Use @rerun-io/web-viewer@0.22.1 in a Quasar + Vite project.
2. Run quasar build and deploy the generated dist/spa/.
3. Open the browser console and observe Rerun failing to fetch .wasm due to a malformed URL.
4. However, the .wasm file is actually present and accessible if requested manually.

**Expected behavior**

- Normal display of the Rerun viewer in the app (as is the case if using rerun 0.16.1). In other words, Rerun should dynamically determine the correct .wasm path in Vite-based production builds.

**Screenshots**

Error as exposed in the dev console:

[image]

which points to:

[image]

**Backtrace**

```
rerunStore-fFXX2zbC.js:1 Uncaught (in promise) TypeError: Failed to construct 'URL': Invalid URL
    at qt (rerunStore-fFXX2zbC.js:1:18593)
    at xt (rerunStore-fFXX2zbC.js:1:18746)
    at fe.start (rerunStore-fFXX2zbC.js:1:20474)
    at async Proxy.create (rerunStore-fFXX2zbC.js:23:667)
    at async IndexPage0-8Kc_m0xt.js:22:150003
```

carueda, 2025-03-04:
BTW, for development mode I previously also faced a vite related issue, then finding https://github.com/rerun-io/rerun/issues/6815, whose indicated workaround is being effective.

jprochazk, 2025-03-06:
I believe the issue here is Vite's lack of proper support for Wasm. We are tracking that here:

* #6815

> Rerun seems to use `import.meta.url` incorrectly inside a `new URL()` call

We need our library to be portable across bundlers, and that is the only way to get the URL of an asset that's even somewhat close to standard. Unfortunately, Vite is the only bundler that doesn't handle this correctly. There is also `import.meta.resolve`, but support for that is nonexistent. Note that we do not directly import the `.wasm` file, because Vite doesn't handle that without a plugin, either.

Can you clarify whether the workaround in #6815 helps with your problem? If not, then we'll need more details. The value emitted in place of `import.meta.url` contains something that is not a valid base URL. I can't investigate as I don't have access to your project - could you dig deeper here and let us know what `import.meta.url` is being resolved to? Thanks!

carueda, 2025-03-06:
Thanks for the quick reply.

The workaround described in #6815 helps but only in _development_ mode, not for a _production_ build.

Sorry, I involved some pieces specific to my project but wanted to bring attention to an issue that seems general with Vite. Unfortunately, my project is not (yet) open source. 

In the minified assets I'm not seeing any occurrence of `import.meta.url`, but I assume it has become `yi.url`.

`rerunStore-DDtJh50d.js` in particular, is the minified script created in my production build, which includes my (Vue3/Pinia) store, which in turn includes Rerun code.

I added this there:
```js
console.debug("yi=", yi, ", yi.url=", yi.url);
```
right before `return fetch(new URL(...`. Here's what I get:

[image]

jprochazk, 2025-03-06:
I believe that needs to be reported upstream to either the (meta-?) framework that you are using, or to Vite. There likely isn't anything we can do on our side.

carueda, 2025-03-06:
OK, feel free to close or keep this open as you deem appropriate.

jprochazk, 2025-03-06:
We'd appreciate it if you could link to any issues you create, if you decide to create them upstream. Thank you for your patience!

closed 2025-03-06 by jprochazk

carueda, 2025-03-06:
@jprochazk - Apologies, despite my original intention, I managed to overlook some relevant adjustments in my project's setup (which uses the Quasar framework, where the vite configuration is done indirectly via some other resource, and I now note they have recent upgrades involving vite).  I'm not seeing this issue after applying those adjustments. Again, thanks for your time.