rerun-io/rerun #9198: Aligning on the design of chunk fetching APIs

teh-cmc, 2025-03-05:
Probably the most important job of the storage APIs is to expose ways of finding and retrieving chunks, so here's a dedicated issue just for those APIs.

Our high-level needs are simple:
* We want callers to be able to fetch chunks from a set of chunk IDs they have lying around.
* We want callers to be able to find sets of chunk IDs using complex, expressive queries.

In practice, this comes down to 3 endpoints:
* `GetChunks`
* `QueryManifest`
* `GetRelevantChunks`

### `GetChunks`

The most basic, low-level RPC: pass chunk IDs in, get chunks out.

How you get your chunk IDs is up to you, though in many cases they will probably come from `QueryManifest`.

```proto
rpc GetChunks(GetChunksRequest) returns (stream GetChunksResponse) {}

message GetChunksRequest {
  CatalogEntry entry = 1;
  rerun.common.v0.RecordingId recording_id = 2;
  repeated rerun.common.v0.Tuid chunk_ids = 3;
}

message GetChunksResponse {
  rerun.common.v0.RerunChunk chunk = 1;
}
```

Notes:
* This is extremely close to what we have today (I just added the dedicated response type).
* Maybe someday this can retrieve chunks using some key other than `ChunkID`, who knows. But generally we expect this to be a simple key-value kind of endpoint.
* There is the eternal question of "should one be able to query a chunk just with a chunk ID (i.e. no need for collection/catalog/resource ID)?"
  * I don't know the answer to that question, but in any case this doesn't fundamentally change the API (although it might or might not change the underlying indices).

### `QueryManifest`

`QueryManifest` is how you find chunk IDs of interest.

You can pass it a latest-at, range or dataframe query (or all three at once!), and it will return a bunch of metadata about all the chunks needed to perform these queries (including, but not limited to, their chunk IDs).

The ability to pass multiple queries at once is what makes things like URL sharing possible: "hey, im looking for chunks ranging from T0 to T1 and also for latest-at chunks at T-1".
Maybe one day all we'll need is the dataframe API, but today is not that day.

```proto
rpc QueryManifest(QueryManifestRequest) returns (stream QueryManifestResponse) {}

// A manifest query will find all the relevant chunk IDs (and optionally a bunch of related metadata)
// for a given set of Rerun queries (latest-at, range, dataframe).
//
// The result might contain duplicated chunk IDs, it is the responsibility of the caller to deduplicate
// them as needed.
message QueryManifestRequest {
  CatalogEntry entry = 1;
  rerun.common.v0.RecordingId resource_id = 2;

  // What columns of the manifest are we interested in?
  ColumnProjection columns = 3;

  ManifestQueries queries = 4;

  // Generic parameters that will influence the behavior of the Lance scanner.
  ScanParameters scan_parameters = 5;
}

message QueryManifestResponse {
  DataframePart data = 1;
}

message ManifestQueries {
  // If specified, will perform a latest-at query with the given parameters.
  ManifestQueryLatestAtRelevantChunks latest_at = 1;

  // If specified, will perform a range query with the given parameters.
  ManifestQueryRangeRelevantChunks range = 2;

  // If specified, will perform a dataframe query with the given parameters.
  ManifestQueryRangeRelevantChunks range = 3;
}

message ManifestQueryLatestAtRelevantChunks {
  // Which entity paths are we interested in?
  repeated rerun.common.v0.EntityPath entity_paths = 10;

  // Which index column should we perform the query on? E.g. `log_time`.
  rerun.common.v0.IndexColumnSelector index = 20;

  // What index value are we looking for?
  int64 at = 30;

  // Which components are we interested in?
  //
  // If left unspecified, all existing components are considered of interest.
  //
  // This will perform a basic fuzzy match on the available columns' descriptors.
  // The fuzzy logic is a simple case-sensitive `contains()` query.
  // For example, given a `log_tick__SeriesLine:StrokeWidth#width` index, all of the following
  // would match: `SeriesLine:StrokeWidth#width`, `StrokeWidth`, `Stroke`, `Width`, `width`,
  // `SeriesLine`, etc.
  repeated string fuzzy_descriptors = 40;
}

message ManifestQueryDataframeRelevantChunks {
    // TODO: not implemented
}
```

Notes:
* This is very close to what we have today.
* We only support range queries today (and badly). Need the rest ASAP.
* Whether a manifest covers 1 or N recordings is still a pending question. Doesn't change much though, one can easily extrapolate from here.
* There is the eternal question of "should one be able to query a resource just with a resource ID (i.e. no need for collection/catalog)?"
  * I don't know the answer to that question, but in any case this doesn't fundamentally change the API (although it might or might not change the underlying indices).
* In general, there is a lot of pending questions around the whole manifest data model still.

### `GetRelevantChunks`

`GetRelevantChunks` is just a helper that combines `QueryManifest` and `GetChunks` because A) it is a very common need and that makes it much more convenient to use and B) it's more performant to keep everything streaming on the server-side.

```proto
rpc GetRelevantChunks(GetRelevantChunksRequest) returns (stream GetRelevantChunksResponse) {}

// A manifest query will find all the relevant chunk IDs (and optionally a bunch of related metadata)
// for a given Rerun query (latest-at, range, dataframe).
//
// The result might contain duplicated chunk IDs, it is the responsibility of the caller to deduplicate
// them as needed.
message GetRelevantChunksRequest {
  CatalogEntry entry = 1;
  rerun.common.v0.RecordingId resource_id = 2;

  ManifestQueries queries = 3;

  // Generic parameters that will influence the behavior of the Lance scanner.
  ScanParameters scan_parameters = 4;
}

message GetRelevantChunksResponse {
  rerun.common.v0.RerunChunk chunk = 1;
}
```

jleibs, 2025-03-05:
Thanks for the write-up. The idea of "convenience gRPC calls" still seems a bit weird to me.

I would have expected that kind of functionality to be implemented in the helper libraries that sit on top of the gRPC calls so that we could always go through a uniform "client-side chunk cache".

jleibs, 2025-03-06:
> That said, there is also some inherent value to having all-in-one endpoints

I'm sold. Maybe let's just make sure to document them as such to make sure we don't accidentally use them in places we shouldn't.

closed 2025-07-11 by teh-cmc