rerun-io/rerun #9215: Project: User-defined recording properties

grtlr, 2025-03-06:
## Summary description

Enable users to assign custom properties to a recording, as well as registering currently hardcoded information (such as `application_id`, name of a recording) late during the logging process.

## Background and problem

It is very common for the user to have properties (metadata) that they want to associate with a recording—some of this information might only be available at the end of the recording. Examples of this could be _"How was the weather?"_, or _"Was the task in this episode executed correctly?"_.

Right now, we only provide very limited, or hardcoded bits of information, such as `application_id` and `started` time. It's important to note that the this aforementioned information is not part of any Arrow `RecordBatch` making it inaccessible for the dataplatform to index.

## Grounding use-cases

1. The user wants to assign a name (such as _"Episode 42"_) to the recording that shows up in the viewer instead of the timestamp that we currently use as an item below the application id in the recording panel.
2. The user wants to gain quick insights, such as camera information, from a recording by hovering over a recording name (without the need for switching to that recording).
3. The user wants to assign tags to recordings (like `rainy`, `cloudy`, ...) to a recording. The dataplatform will be able to pick up these tags and show them as properties in a catalog view, and use them to create derived views (all recordings where weather was `rainy`) to feed into training pipelines.
4. The user can assign complex information (semantic information that exceeds string-based key-value pairs), such as thumbnails, to a recording directly from the SDK, something that does not work via the dataplatform CLI and currently requires implementing a custom gRPC client.
5. The users can evolve their recording properties over time, adding additional information when needed, while maintaining backwards compatibility.

## How to describe, demo and evaluate

* Recording properties show up in the selection panel and in tooltips.
* Show how users can query the dataplatform for interesting recordings.
* Large datasets (such as LeRobot) in the _Recording Panel_ will look much more organized recording names
* Demo how users can rename recordings live via the logging SDK.

## Designs and plans

Overall, the goal is to move as much information as possible to Sorbet-compliant record batches that the dataplatform directly knows how to handle (for querying and indexing). That means we have to move over information that is currently sent via gRPC headers (`StoreInfo`) into `Chunks`.

Another big advantage of moving information to the `Chunks` is that we can evolve their definitions via Sorbet's backwards compatibility mechanisms (#backwa

## Tasks

- [ ] Simple inline tasks
- [ ] Links to sub-issues

## Non-goals and won't do

grtlr, 2025-04-28:
This has been implemented except for #9193 — which we can revisit separately.

jleibs, 2025-04-30:
From "How to describe, demo and evaluate"
> Show how users can query the dataplatform for interesting recordings.

This isn't done yet.
- https://github.com/rerun-io/dataplatform/issues/652