rerun-io/rerun #9244: Allow rotation of all logged entities in Spatial3D view

pablovela5620, 2025-03-09:
**Is your feature request related to a problem? Please describe.**

When logging with a slam method, the output is slightly rotated such that it has a 45 degree tilt, this is annoying to fix currently as I would need to log a rotation to the parent path and do it someone guess and check

**Describe the solution you'd like**

Ideally, there would be some gui/viewer way to rotate all logged entities similar to other overrides

**Describe alternatives you've considered**

Logging the rotation directly via code to the parent log path, sometimes this works, other times its more complex and doesn't work as expected

**Additional context**

Example of what I mean, and how having some rotation in the GUI would simplify things (rotated relative to the ground plane)

[image]

Wumpf, 2025-03-10:
> Logging the rotation directly via code to the parent log path, sometimes this works, other times its more complex and doesn't work as expected

Can you add some more words to this? When is it complex & doesn't do what you expect? Seems like something that should also be addressed

Otherwise closely related to:
* https://github.com/rerun-io/rerun/issues/6743

-> If it was possible to override rotations in the ui everywhere, we could also rotate a space's origin

pablovela5620, 2025-03-11:
Sure, I have two examples that you can find in the mast3r slam repo

[example 1 link](https://github.com/rerun-io/mast3r-slam/blob/53827620cd4482e03ab49f9e5944e53ac921b5c6/tools/load_ns_studio.py#L22)[
example 2 link
](https://github.com/rerun-io/mast3r-slam/blob/53827620cd4482e03ab49f9e5944e53ac921b5c6/mast3r_slam/api/inference.py#L59)
```python
def view_ns_data(config: ViewNsDataConfig) -> None:
    with open(config.transform_json_path, "r") as f:
        ns_data: NerfstudioData = from_json(NerfstudioData, f.read())

    ply_path: Path = config.transform_json_path.parent / ns_data.ply_file_path
    pcd = o3d.io.read_point_cloud(str(ply_path))

    pcd = pcd.voxel_down_sample(voxel_size=0.03)

    points = np.asarray(pcd.points)
    colors = np.asarray(pcd.colors)

    parent_log_path = Path("world")

    rr.log(f"{parent_log_path}", rr.ViewCoordinates.RDF, static=True)
    rr.log(
        f"{parent_log_path}/point_cloud",
        rr.Points3D(positions=points, colors=colors),
        static=True,
    )

    # Apply the rotation to the root coordinate system
    rr.log(
        f"{parent_log_path}",
        rr.Transform3D(
            rotation=rr.RotationAxisAngle(axis=(0, 0, 1), radians=-np.pi / 4)
        ),
        static=True,
    )
```

in the above case the rotation is applied as expected and there's a 45 degree rotation
---

```python
def mast3r_slam_inference(inf_config: InferenceConfig):
    mp.set_start_method("spawn")
    torch.backends.cuda.matmul.allow_tf32 = True
    torch.set_grad_enabled(False)
    device = "cuda:0"

    ## rerun setup
    parent_log_path = Path("/world")
    rr_logger = RerunLogger(parent_log_path)
    # create a blueprint
    blueprint: rrb.Blueprint = create_blueprints(parent_log_path)
    rr.send_blueprint(blueprint)

    load_config(inf_config.config)
    print(inf_config.dataset)
    print(config)
```

```python
class RerunLogger:
    def __init__(self, parent_log_path: Path):
        self.parent_log_path: Path = parent_log_path
        # Create a 3x3 rotation matrix for 90-degree rotation around X-axis
        rr.log(f"{self.parent_log_path}", rr.ViewCoordinates.RDF, static=True)
        # this does not work and I don't know why
        rr.log(
            f"{parent_log_path}",
            rr.Transform3D(
                rotation=rr.RotationAxisAngle(axis=(0, 0, 1), radians=-np.pi / 4)
            ),
            static=True,
        )

        self.path_list = []
        self.keyframe_logged_list = []
        self.num_keyframes_logged = 0
        self.conf_thresh = 7
        self.image_plane_distance = 0.2
```
In this case, no rotation is applied and its not clear why (even though I'm doing basically the exact same thing)
---

pablovela5620, 2025-03-21:
another slightly different example where having the option to rotate all entities after logging

https://github.com/user-attachments/assets/f740ba25-2fb2-4ca5-af90-59bf201cb57f