rerun-io/rerun #9280: Add single timestamp to query parameters in `re_uri`

grtlr, 2025-03-13:
Right now we can only encode a time range (`from` and `to`) into our URIs, but we should also be able to encode a single timestamp to navigate to.

Adding the query parameter to `re_uri` is easy, the more tricky parts lie in the viewer/dataplatform, where the range query, needs to be changed to a `LatestAtQuery`.

teh-cmc, 2025-03-14:
Support for all standard queries in the manifest is in progress, see:
* https://github.com/rerun-io/dataplatform/issues/316
* https://github.com/rerun-io/dataplatform/issues/317
* https://github.com/rerun-io/dataplatform/issues/318

Wumpf, 2025-04-28:
We still don't have anything that does a latest-at query for querying the data. What we do have is jumping to a specific timepoint, but that's different from the query setting

emilk, 2025-04-28:
To clarify:
This issue is about adding a `?query` parameter, controlling what the backend returns.
https://github.com/rerun-io/rerun/pull/9535 added a `#fragment` for setting the viewer's time cursor.

These are orthogonal things. `?query` is used by the backend, and `#fragment` by the frontend (viewer).