rerun-io/rerun #9345: External DataLoader support for Web Viewer

methylDragon, 2025-03-21:
Currently there is no way to support loading of files into external dataloaders from the WebViewer.

It would be good if there was such support, perhaps via sending data to the Websocket server or the process that is hosting the web viewer to then send to the Websocket server.

Perhaps one or more of the following could be implemented:
- Paired with https://github.com/rerun-io/rerun/issues/5324 , have the WebViewer send or contact the data from the file over any of the transport to load in
- Send the file contents over transport (gRPC?) to the hosting process
- Send the file contents over via the same Websocket connection tied to the server
- Send the file path instead
  - This last one is probably not the best because it would require that the web viewer is located in the same host
- Add a new UI to the viewer to specify the URL of a file to load, and send that instead

siddk-lbm, 2025-09-09:
This feature would be incredible; just wanted to echo our team's interest in having this functionality as well.

To the extent that the core Rerun team can give high-level pointers on the best way to implement this functionality, we might also be able to contribute some developer cycles to help out!

Wumpf, 2025-09-09:
Dataloaders are executed on the SDK side therefore this actually already works on the web. What doesn't is to open files on your web viewer since the web can't access the (currently only locally installed) data loaders.

For details see https://rerun.io/docs/reference/data-loaders/overview

@methylDragon come to think is the original issue actually about loading directly in the web, or which part exactly is missing for you?

methylDragon, 2025-10-05:
> Dataloaders are executed on the SDK side therefore this actually already works on the web. What doesn't is to open files on your web viewer since the web can't access the (currently only locally [...]

I was thinking something like, from the **web viewer** UI, if a user drops in a file that we can load it with a custom external DataLoader.

Ideally this triggers the same paths as calling `log_file` from the SDK, but at the time of me writing this issue, it did not.