rerun-io/rerun #9355: Color alpha value doesn't work as expected in Boxes3D and Mesh3D

mingyu623, 2025-03-21:
**Describe the bug**

The alpha channel in rerun::Color doesn't work as expected in Boxes3D or Mesh3D.

**To Reproduce**
Steps to reproduce the behavior:
```
int main() {
  const auto rec = rerun::RecordingStream("boxes3d_exp");
  rec.spawn();

  const rerun::HalfSize3D halfSize(0.5, 1.0, 0.01);
  std::vector<rerun::Color> alphaColors = {
    {255, 0, 0, 8}, {255, 0, 0, 64}, {255, 0, 0, 128}, {255, 0, 0, 255}};

  rec.log(
      "boxes",
      rerun::Boxes3D::from_centers_and_half_sizes(
          {{1,0,0}, {1.8,0,0.2}, {2.6, 0, 0.4}, {3.4,0,0.6}},
          {halfSize, halfSize, halfSize})
          .with_colors(alphaColors)
          .with_fill_mode(rerun::FillMode::Solid));
}

```
**Expected behavior**

The boxes should be from right to left should be more transparent as the alpha value goes down.

**Screenshots**

The boxes go darker from right to left as the alpha value goes down. And you can clearly see they're opaque. 
[image]

**Rerun version**

0.20.0

**Additional context**

Similar issue happens to Mesh3D, which doesn't even take the alpha value, meaning the rendered mesh doesn't change with the input alpha value.

mingyu623, 2025-03-21:
Here's the code snippet to repro the Mesh3D alpha issue.
```
    std::vector<rerun::Position3D>& vertices;
    std::vector<rerun::TriangleIndices>& triangleIndices;
    std::vector<rerun::Color>& colors;

    vertices.push_back({0, 0, 0});
    vertices.push_back({1, 0, 0});
    vertices.push_back({0, 1, 0});
    colors.insert(colors.end(), 3, {255, 0, 0, 8});

    vertices.push_back({0, 0.5, 0.1});
    vertices.push_back({1, 0.5, 0.1});
    vertices.push_back({0, 1.5, 0.1});
    colors.insert(colors.end(), 3, {255, 0, 0, 128});

    vertices.push_back({0, 1, 0.2});
    vertices.push_back({1, 1, 0.2});
    vertices.push_back({0, 2, 0.2});
    colors.insert(colors.end(), 3, {255, 0, 0, 255});

   rec.log(
      "triangles",
     rerun::Mesh3D(vertices).with_vertex_colors(colors));
```
From the screenshot, it shows that the alpha value doesn't even do anything for vertex color.
[image]

series8217, 2025-03-28:
Not working with Boxes2D in a Spatial2DView either. It just gets darker instead.

emilk, 2025-03-30:
This is because we haven't implemented transparency yet:
* https://github.com/rerun-io/rerun/issues/1611

closed 2025-03-30 by emilk