rerun-io/rerun #9368: Shader reloading features cause viewer to crash when visualizating tensors (Bazel-built viewer only)

cygann, 2025-03-24:
**Describe the bug**
When running a rerun application built with Bazel, it will crash when attempting to visualize a tensor:

```
thread 'ThreadId(16)' panicked at 'called `Result::unwrap()` on an `Err` value: include_file!(../../shader/instanced_mesh.wgsl) (rooted at "external/crate_index__re_renderer-0.21.0/src/renderer/mesh_renderer.rs") failed while trying to import physical path "/private/var/tmp/_bazel_nataliec/c4592b75718af3c30c331df93ad7485e/sandbox/darwin-sandbox/3/execroot/rSW/external/crate_index__re_renderer-0.21.0/external/crate_index__re_renderer-0.21.0/src/renderer/../../shader/instanced_mesh.wgsl"

Caused by:
    No such file or directory (os error 2)
```

Upon inspecting the `file_server.rs` implementation, this is caused by path localization & stripping that is incompatible withe Bazel. 

I have unblocked myself by applying the attached patch to disable shader reloading (as this is not a feature I require), but ultimately I am reporting in the interest of one day deleting patches in my workspace so updating rerun is easy in the future. Everything else works as expected after the patch.

[fix_path_re_renderer copy.patch](https://github.com/user-attachments/files/19438851/fix_path_re_renderer.copy.patch)

There may be an elegant way to implement shader reloading to work well with Bazel. Short of that, I wonder if the following would be an appropriate solution:
1) A crate flag that disables shader reloading everywhere and
2) Make the `cfg(not(load_shaders_from_disk))` case not do any path stripping that makes assumptions about where the shader files are. From my understanding of the `load_shaders_from_disk` case, the virtual file system should not require knowledge of the actual paths the shaders live at (i.e. (`CARGO_MANIFEST_DIR`, `file!()`) since `workspace_shaders.rs` loads them at compile time.

Let me know your thoughts!

**To Reproduce**
1. Add rerun packages to a crates_repository
2. Create a `rust_binary` target that runs any example script to launch the visualizer
3. Log a tensor object to rerun
4. Boom

**Expected behavior**
Tensor should visualize in the viewer & it should not crash

**Backtrace**
See above

**Desktop (please complete the following information):**
 - MacOS Sequoia 15.2

**Rerun version**
Occurs on both 0.21.0 and 0.22.1

emilk, 2025-03-25:
Hot shader reloading is something that we Rerun developers _occasionally_ use when developing shaders. Having it enabled for all debug builds makes very little sense today.

I think we can make this feature opt-in instead. For instance, add an opt-in cargo feature to re_renderer, that could also be triggered with `RERUN_HOT_SHADERS=1` via `re_renderer/build.rs`, or something along those lines.

I've added it to the milestone for the next release.

emilk, 2025-03-27:
This should be solved by:
* https://github.com/rerun-io/rerun/pull/9414

closed 2025-03-28 by emilk