rerun-io/rerun #9380: Python API: add useful deprecation decorator

emilk, 2025-03-26:
The default `@deprecated` decorator in Python has several downsides:

A) it is completely silent, unless users _opt-in_ to them using  `import warnings; warnings.filterwarnings("always")`
B) using deprecated doesn't stop our CI from being green

So, I want a `@rerun_deprecated` decorator that solves this:

A) _always_ prints a warning to stderr
B) fails if `RERUN_STRICT=1` is set
C) requires a `since=` argument to set which version we deprecated it at

abey79, 2025-03-26:
I was about to suggest that ruff might be an answer to (B), but apparently not yet:
- https://github.com/astral-sh/ruff/issues/14221

emilk, 2025-03-26:
We're doing this instead:
* https://github.com/rerun-io/rerun/pull/9384