rerun-io/rerun #9405: ERROR wgpu::backend::wgpu_core] Handling wgpu errors as fatal by default

Peeta586, 2025-03-27:
**Describe the bug**
```shell
ibEGL warning: DRI3: Screen seems not DRI3 capable                                                                                                                                                           
libEGL warning: DRI2: failed to authenticate                                                                                                                                                                  
[2025-03-27T13:00:27Z ERROR wgpu::backend::wgpu_core] Handling wgpu errors as fatal by default                                                                                                                
                                                                                                                                                                                                              
thread 'main' panicked at 'wgpu error: Validation Error                                                                                                                                                       
                                                                                                                                                                                                              
Caused by:                                                                                                                                                                                                    
  In Surface::configure                                                                                                                                                                                       
    Invalid surface                                                                                                                                                                                           
                                                                                                                                                                                                              
'                                                                                                                                                                                                             
wgpu-24.0.1/src/backend/wgpu_core.rs:3432                                                                                                                                                                     
stack backtrace:                                                                                                                                                                                              
   6: core::panicking::panic_fmt                                                                                                                                                                              
   7: wgpu::backend::wgpu_core::ContextWgpuCore::handle_error_inner                                                                                                                                           
   8: <wgpu::backend::wgpu_core::CoreSurface as wgpu::dispatch::SurfaceInterface>::configure                                                                                                                  
   9: wgpu::api::surface::Surface::configure                                                                                                                                                                  
  10: egui_wgpu::winit::Painter::configure_surface                                                                                                                                                            
  11: egui_wgpu::winit::Painter::resize_and_generate_depth_texture_view_and_msaa_view                                                                                                                         
  12: pollster::block_on                                                                                                                                                                                      
  13: <eframe::native::wgpu_integration::WgpuWinitApp as eframe::native::winit_integration::WinitApp>::resumed                                                                                                
  14: <eframe::native::run::WinitAppWrapper<T> as winit::application::ApplicationHandler<eframe::native::winit_integration::UserEvent>>::resumed                                                              
  15: winit::platform_impl::linux::x11::EventLoop<T>::run_on_demand                                                                                                                                           
  16: eframe::native::run::run_wgpu                                                                                                                                                                           
  17: eframe::run_native                                                                                                                                                                                      
                                                                                                                                                                                                              
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting 

```

## environment
```shell
python=3.10
lerobot=latest
rerun-sdk=0.22.1
```

Wumpf, 2025-03-31:
@Peeta586 What system are you running on? GPU(s), OS would be relevant. I can see from the log output already that you're running on Linux with X11, correct?
You get this crash presumably on every startup via python?
Also it might be helpful to get some additional log output - can you try running with `RUST_LOG=trace` and post that here?

Forcing either gl or vulkan via `--renderer=gl` or `--renderer=vulkan` might be interesting as well to try out.

---

Iirc upcoming wgpu versions will add a little bit more for this error message. But we should handle this a bit better, we're clearly installing wgpu error handlers too late (this still happens inside egui, re_renderer then later adds additional error handlers)

Wumpf, 2025-06-10:
@ShenhanQian Looks like you have both an intel and an nvidia gpu but for some reason the intel one ends up being picked and then fails to create a surface to present to.
```
[2025-06-10T07:05:26Z INFO  egui_wgpu] There were 3 available wgpu adapters: {backend: Vulkan, device_type: IntegratedGpu, name: "Intel(R) Graphics (RKL GT1)", driver: "Intel open-source Mesa driver", driver_info: "Mesa 24.2.8-1ubuntu1~24.04.1", vendor: Intel (0x8086), device: 0x4C8A}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 19.1.1, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 24.2.8-1ubuntu1~24.04.1 (LLVM 19.1.1)", vendor: Mesa (0x10005)}, {backend: Gl, device_type: Other, name: "NVIDIA RTX A4000/PCIe/SSE2", driver_info: "3.3.0 NVIDIA 560.35.03", vendor: NVIDIA (0x10DE)}
[2025-06-10T07:05:26Z DEBUG egui_wgpu] Picked wgpu adapter: backend: Vulkan, device_type: IntegratedGpu, name: "Intel(R) Graphics (RKL GT1)", driver: "Intel open-source Mesa driver", driver_info: "Mesa 24.2.8-1ubuntu1~24.04.1", vendor: Intel (0x8086), device: 0x4C8A
```

Also this part is interesting:
```
Can't open bumblebee display.
```

Can you try starting with `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/nvidia.json` to force the nvidia gpu?
Do other vulkan applications like vkcube (usually pre-installed) work for you?

Wumpf, 2025-06-10:
thanks for responsiveness & logs in turn! `vkcube` failing implies that something with your vulkan setup is broken. A bit hard to diagnose, linux multi gpu & driver setups are always a bit tricky and I'm not running Linux myself. Other folks online running into `Can't open bumblebee display.` & similar are proposing all sort of solutions/workarounds, not sure what ends up actually working 🤷 

Given that glxgears works Rerun's gl backend is likely to work so try running with `--renderer gl` and see what output that gives :)

Note that in the long run we'd like to deprecate our gl backend and only use Vulkan.

Wumpf, 2025-06-10:
actually the output from the previous log implies that it also struggles with the gl backend and now didn't find any vulkan adapter at all. Strange!

Wumpf, 2025-06-10:
awesome!!

Given that @Peeta586 never responded with a log and you were able to resolve a similar issue. I'm closing this.
Anyone hitting something like this again please open a new issue and link this for reference if you think it's related!

closed 2025-06-10 by Wumpf