rerun-io/rerun #9406: Make C++ build fail if using a deprecated function

emilk, 2025-03-27:
If we're using a function marked `[[deprecated]]` in any of our examples or snippets, that should turn the CI red.

ishakbhatt, 2025-04-08:
Taking a look at this!

emilk, 2025-04-09:
Thank you! I did add a few compiler flags to our top-level `CMakeLists.txt`: https://github.com/rerun-io/rerun/blob/main/CMakeLists.txt#L56 but I'm not sure they catch everything.

In particular, we want `pixi run -e cpp cpp-build-all` to fail if any of our examples or snippets (`docs/snippets/**/*.cpp`) use a `class` or function marked `[[deprecated]]`. Maybe it already works, but needs proper verification.

ishakbhatt, 2025-04-10:
Status update: 

I attempted to verify that `Werror=deprecated-declarations` fails snippets or examples (I started with `arrows3d_row_updates`) and I am seeing that I can still call deprecated functions without errors.

It seems, after printing compiler flags with what is in `main`, that it's only printing compiler flags from pixi, which is generated in `.pixi` in `.pixi/envs/cpp/include`. I went ahead and changed it to `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wdeprecated -Werror=deprecated-declarations")` in the main `CMakeLists.txt` and it shows up. It is not ideal for users to have to do the same in their projects, but I am also investigating doing the same in the other `CMakeLists.txt`. 

I will make another comment here when it is fully verified! Just wanted to give an update here on what I investigated.

closed 2025-04-24 by timsaucer