rerun-io/rerun #9433: First person view 0 movement speed

szaboboldizsarmark, 2025-03-31:
**Unable to move in first person (orbit camera can move) when the scene is infinitely small (e.g. only consists of a single point**

---

**The issue**
This is an oversight, rerun calculates the orbit camera movement speed based on the zoom level, but since the first person view camera doesn't zoom in or out, a different approach had to be taken, where the movement speed of the camera is based on the scene size:

Small scene size -> small movement speed
Big scene size -> a lot of movement speed

When the scene size is infinitely small (e.g. scene that only has a single Points3D point), the movement speed of the first person camera will be 0, thus you are unable to move. Having your scene consisting of a single point is actually something that you might want to do, if you are just visualizing a position for example (this is how I discovered the bug).

---

**Solution**
Calculating the movement speed based off of scene size is a terrible idea in general, what if you had a huge scene but wanted the camera precisely (slowly)? You can't do that, since movement speed is fixed.

If we look back how the orbit camera handles the speed calculation we can see that we use the mouse wheel to adjust the zoom which intern adjusts the movement speed. When using the first person camera the mouse wheel actually doesn't do anything (since zooming in is only relevant in orbit camera), why not just let the user adjust the movement speed with the mouse wheel? This would solve all of our issues.

Now we can:
- Fine tune our movement speed to enable slow and precise movements (just like with the orbit camera), while still letting us increase it to move great distances when our scene is large
- Completely solves the 0 movement speed bug when the scene is only a single point.

Note:
Alongside making the movement speed adjustable with the mouse wheel, I would also keep the current calculation, but instead use it to figure out an initial movement speed, and then the user can adjust if they feel like it. I would also clamp this initial movement speed to a non zero value, to prevent the initial movement speed being zero when the scene is infinitely small.

A lot of video games let you use this type of camera control (adjustable speed) when in spectator mode (e.g. Minecraft, Space Engineers, etc...), so implementing this would make a lot of sense.

---

**To Reproduce**
Steps to reproduce the behavior:
1. Add a single Points3D point to a newly created scene with nothing else in it
2. Switch to the first person camera view
3. Experience not being able to move

**Desktop (please complete the following information):**
 - OS: 

**Rerun version**

**Additional context**

Wumpf, 2025-03-31:
>  why not just let the user adjust the movement speed with the mouse wheel

part of blueprint configurable camera, see 
* https://github.com/rerun-io/rerun/issues/6757

concrete issue of viewer automatically picking a broken speed value will remain regardless because it has to pick _something_ in absence of user input as a start value as you pointed out. -> configurable camera (speed) is a separate problem from broken heuristic described here.
Not sure about camera speed for first person being controlled by mousewheel - without getting immediate feedback on what happens this sounds very unintuitive.

szaboboldizsarmark, 2025-03-31:
@Wumpf I agree, a little widget for this would be perfect!

Wumpf, 2025-08-15:
speed is now configurable and persisted in the blueprint. But looks like the default speed of fps cameras in empty scenes is now infinity. So default still needs fixing