rerun-io/rerun #9521: View heuristics are broken

emilk, 2025-04-07:
The default heuristics are completely broken at the moment. Here is the heuristic blueprint for plots:

[image]

This is what it was in egui 0.22:

[image]

emilk, 2025-04-07:
We should check the heuristics for all other examples before marking this as done

grtlr, 2025-04-10:
Ran `git bisect`, which was a bit tricker because of changes to our file format. I've added a `.zip` file with the required `.rrd` files.

Here are the results:

* The first bad commit seems to be 7995dd4a and requires `commit-7c297e1-examples-plots.rrd`
* The last good commit seems to be e7a96189 and requires `commit-67e110b-examples-plots.rrd`

Here are the files: [plots.zip](https://github.com/user-attachments/files/19684250/plots.zip)

grtlr, 2025-04-10:
Ok, I ran another `git bisect` this time with the exact `.rrd`s per commit using:

```sh
pixi run reset && pixi run rerun "https://storage.googleapis.com/rerun-web-viewer/commit/$(git rev-parse --short=7 HEAD)/examples/plots.rrd"
```

* Start good at 6b5a938
* Start bad at 9d4a590f
* Skipped 0fa8e93 (merge commit)

Result: https://github.com/rerun-io/rerun/commit/d717148db08c9babdaa2a1a2eb4141b51f809c30

Which actually changes the `plots.py`.

grtlr, 2025-04-10:
Probably not introduced by the commit above, but might be related to multi-scalar logging.

Discussion: https://rerunio.slack.com/archives/C041NHU952S/p1744286626081289

emilk, 2025-04-11:
So the new `/spiral` entity makes us hit this special-case:
https://github.com/rerun-io/rerun/blob/acb6f094b7d0bbda5318f22efd1e4397836ffa76/crates/viewer/re_view_time_series/src/view_class.rs#L246-L258

…which seems like a weird heuristic to me. The comment links to https://github.com/rerun-io/rerun/issues/4926

I'll see if I can improve this

closed 2025-04-12 by Wumpf