rerun-io/rerun #9556: A simple demo got incrediable big file size!

OpenJarvisAI, 2025-04-09:
Hi, got a simple demo, records frames from video to rrd

```
import rerun as rr
import sys
import cv2
import os

def main():
    # Read video path from command line
    if len(sys.argv) != 2:
        print("Usage: python video_to_rrd.py <video_path>")
        sys.exit(1)
    
    video_path = sys.argv[1]
    video_path = os.path.expanduser(video_path)
    
    # Initialize Rerun
    # rr.init("video_storage", spawn=True)
    if not os.path.exists(video_path):
        print(f'video not found. {video_path}')
    
    # Open video file
    cap = cv2.VideoCapture(video_path)
    if not cap.isOpened():
        print(f"Error: Could not open video file {video_path}")
        sys.exit(1)
    
    rr.init('video')
    
    # Get actual FPS from video
    fps = cap.get(cv2.CAP_PROP_FPS)
    
    frame_idx = 0
    while True:
        ret, frame = cap.read()
        if not ret:
            break
        
        # Convert BGR to RGB
        frame_rgb = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
        
        # Log to Rerun
        rr.set_time_sequence("frame_time", int(frame_idx))
        rr.log("video/frames", rr.Image(frame_rgb))
        
        frame_idx += 1
    
    # Save and clean up
    cap.release()
    rr.save("video_storage.rrd")
    print(f"Saved {frame_idx} frames to video_storage.rrd")

if __name__ == "__main__":
    main()
```

the video size is 2M, but the rrd file is about 478M, why

teh-cmc, 2025-04-09:
Your code is extracting and decompressing each individual frame from the original video file instead of logging the video itself and decoding it on the fly during visualization.

See [the documentation for how to efficiently work with video](https://rerun.io/docs/reference/video).

closed 2025-04-09 by teh-cmc

Wumpf, 2025-04-10:
Rerun doesn't just show the images you're sending to it, it stores all of them in memory so you can replay it. So essentially you're decoding the entire video and Rerun stores that for you. If you want to keep only the last images around that you log you can log them with `static=True`, but that's more of a workaround. If you actually want to log a compressed video in Rerun, check the link @teh-cmc shared.