rerun-io/rerun #9765: Does rerun involves any method replaying any cpu or memory performance logs?

WorstCodeWay, 2025-04-18:
First of all, I find this project casual, and it is exactly what I want, it's awesome. Thanks for your great works.

**Is your feature request related to a problem? Please describe.**

Sometime in simulate or analysing bugs, we want to know about not only images or clouds or other sensors data, but also cpu or memory consumption。

**Describe the solution you'd like**

If rerun sdk supplies some function to be used to record cpu, memory and maybe other things performance. And then we can replay those data on viewer to show performance statics along with other sensors data. It's very useful in my life to locate a bug.

**Describe alternatives you've considered**

Don't know any alternatives. May be [tracy](https://github.com/wolfpld/tracy.git), but it's too heavy.

**Additional context**

Wumpf, 2025-04-19:
Thanks for the kind words! I agree, it would be pretty neat to easily extract that data and log it to Rerun, renamed the issue accordingly. So far you'd have to gather that data yourself and log it as scalars and/or text log events yourself.

This is also closely related to
* https://github.com/rerun-io/rerun/issues/4631

.. and I think there was the idea floating for logging callstacks with rerun log calls in an opt-in fashion.

WorstCodeWay, 2025-04-20:
> Thanks for the kind words! I agree, it would be pretty neat to easily extract that data and log it to Rerun, renamed the issue accordingly. So far you'd have to gather that data yourself and log it as scalars and/or text log events yourself.
> 
> This is also closely related to
> 
>     * [Spans and flame graphs #4631](https://github.com/rerun-io/rerun/issues/4631)
> 
> 
> .. and I think there was the idea floating for logging callstacks with rerun log calls in an opt-in fashion.

Thank you, @Wumpf. I'm new to rerun, but I'm working on my project integrating rerun. I will try your suggestions first to see what can I get with rerun.