rerun-io/rerun #9783: how to use c language to call rerun_sdk to send data to rerun host computer?

banzhangzz, 2025-04-23:
In esp_idf, we use rerun as the host computer, how to use c language to call rerun_sdk to send data to the host computer?The example in c is not complete.

Wumpf, 2025-04-28:
We don't have a user facing C SDK right now. The [C SDK](https://github.com/rerun-io/rerun/blob/f7f6b70b5caa8c52bcab1a47ad193a0a2975cbb1/rerun_cpp/src/rerun/c/rerun.h) is what the C++ SDK uses internally with serialized arrow data. In theory it's possible to use this interface directly (the C++ SDK does after all ;-)) but not only is it not very user friendly, it's also extremely unsuited for embedded development like any of the ESP SOCs as it drags in all sort of dependencies, uses threads etc. etc.

closed 2025-04-28 by Wumpf

abey79, 2025-04-28:
More generally, we currently don't have any plans to make a MCU compatible rerun SDK. Although it is theoretically possible, it would involve solving to many new problems for this to be realistic. A more sensible path would be to use existing  MCU centric structured logging framework, and have a translation layer running on a regular computer/server.

`uORB`/`ulog` is such an option ([docs](https://docs.px4.io/main/en/middleware/uorb.html)). As it turns out, I have a WIP `ulog` loader PR here here: https://github.com/rerun-io/rerun/pull/9127.

Note to self: one area that could be explored would be to generate uORB message definitions for our archetypes/components 🤔