rerun-io/rerun #9789: Sub-plots within the TimeSeries view

abey79, 2025-04-23:
## Background

Axis-aligned stacks of plot is a widely requested feature for which there is exists no convenient work-around. #7931 proposes a somewhat generic mechanism to "synchronise" axes between multiple views, in hope to benefit from the layout flexibility of our view system. The present issue is a counter-proposal based on the (admittedly somewhat subjective) observation that the view system is too rigid and abstract for this purpose, and that a pragmatic, "sub-plots-within-a-view" approach would provide the same/similar results with lower complexity, both in UX design and code.

## Proposal

The core proposal is to allow multiple plots in a grid _within the same timeseries view_, à la matplotlib's subplots. Currently, a view can only contain a single plot. Also, within these sub-plots, any align axis would automatically be linked and aligned, without a possibility to opt-out (rationale: limit the number of configuration option + use the view system if you want loosely coupled plots).

## Challenges

- Design of the UI to configure the plot view (see proposal below)
- Design of the blueprint APIs (in particular, how the "entity path filter" mini-language fits in the picture
- Implementation:
  - egui_plot needs quite a bit of cleanup work
  - can the multi-plot layout—and alignement!—be done outside of egui_plots?

## UX proposal

The core proposal consist of a "mini-map" of the plot layout where entities can be dragged and dropped to the desired sub-plot.

By default, there is a single subplot, and all entities produced by the entity path filter are assigned to it:

[image]

An entity would be removed by filtering it out from the entity path filter.

The user would click on the + icons to create further rows/column of subplots, and then drag-and-drop entity to dispatch them to the desired sub-plot:

[image]

### Stretch goal 1: assign "weight" to columns or row

[image]

### Stretch goal 2: optional assign an entity one of the X axes

An area would be displayed at the bottom of each column, in which case that entity would be used as abscissa instead of time values. This of course would require some time alignement for all scalars involved, but this is typically the case anyways. For such plot, the time cursor would be displayed as a marker highlighting the corresponding points, as opposed to a vertical line.

[image]

## Related

- https://github.com/rerun-io/rerun/issues/7931
- https://github.com/rerun-io/rerun/issues/8286

abey79, 2025-06-17:
> Adding [figma design](https://www.figma.com/design/72AuvfbssFJeqqnQoJscYT/Katya-s-playground?node-id=5630-8588&t=J2SqZnOtoZyKvDMY-1) drafts for above ideas, so it won't get lost!!

❤️

[image]

rrad5409, 2025-12-24:
Has there been any progress on this?

abey79, 2025-12-25:
> Has there been any progress on this?

Unfortunately not. An adjacent feature was recently added: link X axis range across multiple view. It allows to 'fake' subplots with multiple views. However, the alignment is not perfect due to the vertical legend taking variable space. 

Maybe we should synchronise the left margin as well when this feature is used? This would allow better alignment across vertically stacked views.

[image]

rrad5409, 2025-12-28:
That would be pretty good, I've noticed that myself. For me this issue is quite important, because I'm trying to track graph multiple boolean values (relay states), and having them all in a single graph makes them all overlap and it's impossible to read. My solution has been to offset each line, but it's very hacky and I'd really like to just have `0/1` values and have them be sub-plotted automatically.

[image]