rerun-io/rerun #9807: Display pop-up with the link to migration guide when upgrading rerun

abey79, 2025-04-25:
Some of our users are unaware of our migration guide, which worsen the experience when running into migration issues. We should have some kind of popup with a link upon migrating.

Wumpf, 2025-06-13:
@gavrelina can you think this through a bit? I imagine this could be annoying when opening older recordings. But I figure it's fine if it's just a toast on our notifications.

gavrelina, 2025-06-18:
What'd be a scenario where users face it?
My assumption is: 
- they update rerun to a new version
- run existing rrd/script
- launching viewer and stuff is broken with a notification `You are trying to load an old .rrd file that's not supported by this version of Rerun.` 

For this one indeed — would be an easy win to link to a migration guide page as a CTA
Are there more scenarios?