rerun-io/rerun #9812: Potential regression of slower speed v0.22.1->v0.23.0

makeecat, 2025-04-25:
## Description
The overall logging speed seems to be slower, not sure if it is related to the spawn process of grpc.

## Reproduction
```
## Install dependencies
cargo install hyperfine
cargo install rerun@0.22.1
mv ~/.cargo/bin/rerun ~/.cargo/bin/rerun@0.22.1
cargo install rerun@0.23.0
mv ~/.cargo/bin/rerun ~/.cargo/bin/rerun@0.23.0

## build testbed
git clone git@github.com:makeecat/peng.git peng
git clone git@github.com:makeecat/peng.git peng-prev
## set peng-prev before bumping rerun from 0.22.1 to 0.23.0
cd peng-prev
git checkout 74b8727bac5c49b67677c08c3f47bb7b1676e802
cargo build --release
cd ../peng
cargo build --release
cd ..

## launch rerun 0.22.1
rerun@0.22.1
## launch peng-prev
hyperfine --warmup 10 -r 30 ./peng-prev/target/release/peng_quad

## launch rerun 0.23.0
rerun@0.23.0
## launch peng
hyperfine --warmup 10 -r 30 ./peng/target/release/peng_quad
```

## Result
on MacOS Apple M1, Sequoia 15.4.1:

peng-prev: 987.1ms +- 69.0ms
peng:      1.143ms += 52.0ms

on ROG STRIX15 x64, Ubuntu 22.04:

peng-prev: 812.6ms +- 27.2ms
peng:      842.8ms +- 17.7ms

jprochazk, 2025-04-29:
Hi, I unfortunately can't reproduce the perf regression on any of my machines.

We profiled similar workloads and found some performance regressions in rrd encode due to excessive allocations. I pushed https://github.com/rerun-io/rerun/pull/9826 which does reduces the number of allocations we do to be the same as in 0.22.1, a little better in some cases.

Could you set the rerun dependency in `peng` to point to that git branch, and check if it makes any difference for you?

On my end, even that doesn't yield any real difference in the numbers I'm seeing for your repro:

```
# Mac M2 (Sonoma 14.5)
peng @ 74b8727bac5c49b67677c08c3f47bb7b1676e802 + rerun 0.22.1
  751.5 ms ±  13.6 ms

peng @ main + rerun 0.23.1
  737.1 ms ±  20.5 ms

peng @ main, patched with jan/encode-perf + rerun 0.23.1
  736.6 ms ±  30.0 ms
```
```
# 7950x (running Fedora 41, kernel 6.13)
peng @ 74b8727bac5c49b67677c08c3f47bb7b1676e802 + rerun 0.22.1
  528.0 ms ±  14.8 ms

peng @ main + rerun 0.23.1
  547.9 ms ±  10.9 ms

peng @ main, patched with jan/encode-perf + rerun 0.23.1
  544.0 ms ±  10.7 ms
```

Which makes sense, we only started seeing differences when logging ~hundreds of MiB, and the provided benchmark logs a total of 150 MiB per run. It's likely that there just isn't enough allocation pressure for it to matter, but that depends on a large number of factors...

makeecat, 2025-04-30:
It did improve the performance:

on MacOS Apple M1, Sequoia 15.4.1:

peng @ 74b8727bac5c49b67677c08c3f47bb7b1676e802 + rerun 0.22.1
   812.6 ms ±  27.2 ms
peng @ main + rerun 0.23.1
  1143.0 ms ±  52.0 ms
peng @ main, patched with jan/encode-perf + rerun 0.23.1
  1018.0 ms ± 141.0 ms

jprochazk, 2025-04-30:
That 141ms variance seems really suspicious... Do you think you could provide a flamegraph of that profile?

makeecat, 2025-04-30:
I disabled the multi-threading by changing `config/quad.yaml`: `use_multithreading_depth_rendering` from `true` to `false`. Here is the new result with the flame graph. I turned off all other applications during the benchmark. One more thing I noticed: 0.23.1 seems to occupy more memory than 0.22.1, wondering if that is as expected.

peng @ 74b8727bac5c49b67677c08c3f47bb7b1676e802 + rerun 0.22.1
1.275 s ±  0.005 s

[image]

peng @ main + rerun 0.23.1
1.284 s ±  0.005 s

[image]

peng @ main, patched with jan/encode-perf + rerun 0.23.1
1.285 s ±  0.004 s

[image]

jprochazk, 2025-04-30:
Those numbers seem in line with expectations. The encoding did get a little bit slower because of protobuf and switching to a different Rust `arrow` implementation.

Can I ask for the same thing on your M1 Mac? The difference there was much greater. I'm hoping that with flamegraphs we'll at least be able to see more clearly what's causing the slowdowns, and we can try to optimize that even without being able to directly reproduce the issue.

makeecat, 2025-04-30:
These flame graph are number were reported from MacOS Apple M1, Sequoia 15.4.1. I can get the number on my linux machine later today.

jprochazk, 2025-04-30:
Ah, ok. I assumed it was your linux machine because the numbers were much closer together. That's interesting.

makeecat, 2025-04-30:
on ROG STRIX15 x64, Ubuntu 22.04:

peng @ 74b8727bac5c49b67677c08c3f47bb7b1676e802 + rerun 0.22.1
1.547 s ±  0.042 s

[image]

peng @ main + rerun 0.23.1
1.643 s ±  0.044 s 

[image]

peng @ main, patched with jan/encode-perf + rerun 0.23.1
1.628 s ±  0.024 s

[image]

jprochazk, 2025-04-30:
For some reason, those flamegraphs are missing symbol names. 🤔 was it compiled with debuginfo enabled? (`[profile.release] debug = true`)

makeecat, 2025-04-30:
I failed to obtain the flame graph with debug info on linux, there is something wrong with my perf config. Here is the updated flame graph with debug info on Mac

on MacOS Apple M1, Sequoia 15.4.1:

peng @ 74b8727bac5c49b67677c08c3f47bb7b1676e802 + rerun 0.22.1
1.275 s ±  0.005 s

[image]

peng @ main + rerun 0.23.1
1.284 s ±  0.005 s

[image]

peng @ main, patched with jan/encode-perf + rerun 0.23.1
1.285 s ±  0.004 s

[image]

jprochazk, 2025-05-05:
Current working theory is that the connection is not ingesting data fast enough, because the Viewer is too slow at decoding. There are other things we can optimize E2E if even that isn't enough

emilk, 2025-06-09:
In a test where I encode a bunch of uncompressed 1024² RGBA images (`tests/rust/log_benchmark/src/image.rs`) I found that by far the slowest part is the IPC encoding of arrow data (release build):

[image]

zooming in:

[image]

This suggests that if we simply encode the chunks in parallel instead on N cores, we will get a N-times speedup.

This also gives a likely reason for the recent regression: we switched from `arrow2` to `arrow-rs`.

Perhaps we could also spend some time optimizing `arrow::ipc::writer::StreamWriter`