rerun-io/rerun #9815: Explore GStreamer based video decoding (at least for h264/h265)

Wumpf, 2025-04-28:
Our current ffmpeg exectuable based video decoding is riddled with bugs and performance issues. Several people pointed out to us that it should be relatively easy to _installed gstreamer binaries_ with [gstream-rs](https://gitlab.freedesktop.org/gstreamer/gstreamer-rs) (thus sidestepping most license issues). Unlike the older Rust ffmpeg binding library this is relatively easy to use.

For specific (!!) gstreamer versions, known hardware configuration and renderer, we should be able to even copy directly gpu decoded images to our textures, allowing much faster decoding. Since our decoder interface already works with wgpu textures (in order to support WebCodec) better, we're already in a great spot for this (in fact we only read back hovered pixels for interaction!)

In short, the motivation is:
* better performance - we want to be able to decode & seek several 4k videos
* sidestep various bugs we have with ffmpeg executable playback
* more formats quicker (h265 should be possible with the ffmpeg executable approach but may need some extra work, others will be harder!)

Since we still _have_ the ffmpeg decoder around (no need to remove it if we don't want to) we're perfectly fine with feature restrictions like no b-frame support, which are exceedingly uncommon in robotics streaming cases as they obviously add a lot of latency.

Wumpf, 2025-06-24:
yeah we definitely better video decoding on all native targets no matter what the exact solution is going to be.
So far whenever I experimented with gstreamer I had _a lot_ of issues getting the install to work out on MacOS specifically and the whole Rust integration & compilation experience was quite awful. But not giving up on that path yet but want to explore more.
Vulkan extensions become more promising slowly (but no Mac support and driver support is patchy..) and there's this new ffmpeg binding which might make things easier compared to the existing  https://github.com/ScuffleCloud/scuffle/tree/81ff9a33ef0a840cfc06bcb50496eee0fd4f4ec3/crates/ffmpeg

>  Ubuntu's native viewer wgpu is being used (I could be wrong), 

very much orthogonal, but just to set this right for reference: wgpu is _always_ the library we use for rendering. It in turn layers on top of either WebGPU/WebGL when on the web and GL/Vulkan/Metal/DX12 when on native.

ntjohnson1, 2025-08-07:
Adding a reference here just in case it is relevant for people searching for gstreamer. Someone on the discord started to put together a gstreamer plugin (in C++). https://github.com/RidgeRun/gst-rerunsink looks like it is still in development but I was able to get it built on mac (but no nvidia gpu) and use it to pipe dummy data to a viewer (the viewer launch itself didn't seem to work).