rerun-io/rerun #9849: Fix wasm compilation of Wasm target on macOS

abey79, 2025-04-30:
Currently, a newer version of llvm/clang must be installed on mac, because the Apple-provided one fails building wasm with:

```
failed to build archive at /Users/tsaucer/src/rerun/target_wasm/wasm32-unknown-unknown/release/deps/libpsm-43030b67c35315ca.rlib: LLVM error: section too large
```

We should figure out a way to either fix the build, or make it so that using pixi solves the issue.

emilk, 2025-05-09:
The PR that first caused this problem was:
* https://github.com/rerun-io/rerun/pull/9764

ntjohnson1, 2025-06-23:
Confirmed with git bisect that this has been the case since datafusion was added.

This PR under datafusion appears to work around the issue https://github.com/apache/datafusion/pull/16418

However, IIUC rerun->datafusion->a few layers->[this assembly](https://github.com/rust-lang/stacker/blob/master/psm/src/arch/wasm32.s) -> [this object](http://github.com/rust-lang/stacker/blob/master/psm/src/arch/wasm32.o). Didn't get a chance to go much deeper than that but I wonder if we can assemble that on mac to fix the root cause (since we know things play nice with latest llvm).

commit 08b84645 (2025-08-11): Fix building Web Viewer on macOS (#10828)

### Related
Closes #9849

### What
Basically mac has a fun combination where you can have `clang`/`llvm`
components but not `llvm-ar`. Mac's `ar` pads bytes in way that make
llvm barf. Our pixi env for c++ has the `cxx-compiler` package which has
a more full fledged llvm install. On mac on activation set `AR` to point
to `llvm-ar`. It works on my machine! I'm not sure if we should have a
second person try or if we should setup a test for the web build on mac
just for this weird edge case.