rerun-io/rerun #9855: SorbetColumnDescriptors requires that Schema's be in a particular order

jleibs, 2025-04-30:
The `try_from_arrow_fields` builder depends on a very specific ordering of fields.

See:
https://github.com/rerun-io/rerun/blob/9b8dbe68e64569f1f3f59a6ae0ade916238ea813/crates/store/re_sorbet/src/sorbet_columns.rs#L139-L156

While this ordering makes sense from a performance perspective it violates the "accept what users (or our dataplatform) throws at us" assumptions.

As part of:
- https://github.com/rerun-io/rerun/pull/9854

We introduced a new helper: `try_from_arrow_fields_forgiving` but it would be nice if there behavior were the default unless there's a very specific reason for it being this way.

abey79, 2025-05-01:
The core issue for that is that `SorbetBatch` has two overlapping, but ultimately distinct goal:

- General-purpose wrapper over Arrow record batches, with facilities to de/serialize sorbet metadata
- Stop-gap for the record-batch <-> chunk de/serialisation

The constraint above (and weirdnesses such as `SorbetBatch` having a `entity_path: Option<EntityPath>` stem from the latter.

A suggestion would be to apply a typestate-like pattern, to exploit the overlap but express the difference between these purposes:
- `SorbetBatch<Flex>`: general purpose wrapper
- `SorbetBatch<Strict>`: chunk de/serialisation

emilk, 2025-05-02:
After a discussion with @abey79 we decided we "just" need to make `SorbetBatch` more general, allowing columns in any order, with each column being a

```rs
pub enum ColumnDescriptor {
    RowId(RowIdColumnDescriptor),
    Time(IndexColumnDescriptor),
    Component(ComponentColumnDescriptor),
}
```

and then we move the ordering-constraint of columns into the more strict `ChunkBatch`

closed 2025-05-09 by emilk