rerun-io/rerun #9859: Toggle X view buttons flaky on Mac 2nd monitor, fullscreen

genemerewether, 2025-05-01:
**Describe the annoyance**
The `Toggle X view` buttons are flaky on a Mac (in clamshell mode, closed) when the rerun app is fullscreen on the 2nd monitor / extended display on Macbook Air.

**To Reproduce**
1. Open rerun viewer
2. Move to 2nd monitor / extended display
3. Ensure full screen
4. Click the `Toggle X view` buttons. Most of the time, nothing happens. Occasionally, I can get one of the buttons to react and hide the view. Haven't been able to get the view back by clicking the buttons. The button grey status does correctly reflect whether the view is toggled.

**Expected behavior**
Buttons work always.

**Screenshots**

**Desktop (please complete the following information):**
Both monitors connected over HDMI.

[image]

[image]

genemerewether, 2025-05-01:
https://github.com/user-attachments/assets/c87fb301-a8a1-4806-ae6b-0f8cf0edf99f

abey79, 2025-05-02:
I can reproduce on both of my external monitors, without needing to close my laptop. On the laptop screen, the issue doesn't reproduce. I have the following display arrangement (the laptop screen is bottom right)

[image]