rerun-io/rerun #9862: AnnotationContext for keypoints broken in 0.23.1

pablovela5620, 2025-05-01:
**Describe the bug**

AnnotationContext no longer works on 0.23.1 when trying to log link connections (on notebooks)

**To Reproduce**
Minimal script to reproduce on a jupyter notebook
```python
import numpy as np
import rerun as rr
import rerun.blueprint as rrb

left_joints = np.array([
    [-0.0956699,   0.00638343,  0.00618631],
    [-0.0715802,  -0.0091389,   0.03199915],
    [-0.0519469,  -0.0082476,   0.0556987 ],
    [-0.0297292,  -0.0136805,   0.07022282],
    [-0.0022312,  -0.018095,    0.0909145 ],
    [-0.0075726,   0.00118307,  0.0268723 ],
    [ 0.02510622,  0.00519243,  0.02908936],
    [ 0.04726213,  0.003894,    0.02897524],
    [ 0.0705249,   0.00461197,  0.03302451],
    [-0.0010094,   0.00490446,  0.00282876],
    [ 0.03017318,  0.00676579, -0.0027657 ],
    [ 0.05307782,  0.00551369, -0.0067102 ],
    [ 0.07899282,  0.00614665, -0.0120408 ],
    [-0.0139343,   0.00242601, -0.0204868 ],
    [ 0.0143799,   0.00449301, -0.0255854 ],
    [ 0.03790041,  0.0028049,  -0.0332192 ],
    [ 0.06080423,  0.00734306, -0.0402022 ],
    [-0.0268829,  -0.0035569,  -0.037023  ],
    [-0.0098685,  -0.003495,   -0.0495218 ],
    [ 0.00599835, -0.0041862,  -0.0598537 ],
    [ 0.02189885, -0.0016281,  -0.0701316 ]
])

MEDIAPIPE_IDS = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20]
MEDIAPIPE_ID2NAME = {
    0: 'WRIST',
    1: 'THUMB_CMC',
    2: 'THUMB_MCP',
    3: 'THUMB_IP',
    4: 'THUMB_TIP',
    5: 'INDEX_MCP',
    6: 'INDEX_PIP',
    7: 'INDEX_DIP',
    8: 'INDEX_TIP',
    9: 'MIDDLE_MCP',
    10: 'MIDDLE_PIP',
    11: 'MIDDLE_DIP',
    12: 'MIDDLE_TIP',
    13: 'RING_MCP',
    14: 'RING_PIP',
    15: 'RING_DIP',
    16: 'RING_TIP',
    17: 'PINKY_MCP',
    18: 'PINKY_PIP',
    19: 'PINKY_DIP',
    20: 'PINKY_TIP'
}
MEDIAPIPE_LINKS = ((0, 1), (0, 5), (0, 17), (5, 9), (9, 13), (13, 17), (1, 2), (2, 3), (3, 4), (5, 6), (6, 7), (7, 8), (9, 10), (10, 11), (11, 12), (13, 14), (14, 15), (15, 16), (17, 18), (18, 19), (19, 20))
# Helps provide context for the rerun viewer, to visualize the difference between triangulated and optimized hand joints
def set_pose_annotation_context() -> None:
    rr.log(
        "/",
        rr.AnnotationContext(
            [
                rr.ClassDescription(
                    info=rr.AnnotationInfo(id=0, label="Triangulated Hand", color=(0, 0, 255)),
                    keypoint_annotations=[
                        rr.AnnotationInfo(id=id, label=name) for id, name in MEDIAPIPE_ID2NAME.items()
                    ],
                    keypoint_connections=MEDIAPIPE_LINKS,
                ),
            
            ]
        ),
        static=True,
    )

rr.init("sequence data")
set_pose_annotation_context()
minimal_blueprint = rrb.Blueprint(collapse_panels=True)
rr.log(
    "left joints",
    rr.Points3D(
        np.array(left_joints),
        colors=(0, 255, 0),
        class_ids=0,
        keypoint_ids=MEDIAPIPE_IDS,
        show_labels=False,
    ),
)
rr.notebook_show(width=1000, height=700, blueprint=minimal_blueprint)
```

**Expected behavior**

Should correctly log keypoint links and id annotation

**Screenshots**

0.23.1 output
[image]
0.22.1 output
[image]

**Backtrace**

**Desktop (please complete the following information):**
 - OS: 

```
pablo@pablo-dl-server:~/0Dev/repos/pi0-lerobot$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.2 LTS
Release:        24.04
Codename:       noble
```

**Rerun version**

```
rerun-cli 0.23.1 (default map_view nasm native_viewer release web_viewer) [rustc 1.84.0 (9fc6b4312 2025-01-07), LLVM 19.1.5] x86_64-unknown-linux-gnu, built 2025-04-26T18:13:37Z
Video features: av1 default ffmpeg nasm serde
```
**Additional context**

pablovela5620, 2025-05-01:
Weirdly enough, if I do things through colab, I don't have this issue at all 🤔 

https://colab.research.google.com/drive/1EioO43vQyoLS2OzHMusYbJD4KIbIsuAy?usp=sharing

Wumpf, 2025-05-05:
Turns out the issue is that we're partially incompatible with numpy 1.x.x
we're using the `copy` argument of `asarray` which isn't available prior to numpy 2
https://numpy.org/doc/2.2/reference/generated/numpy.asarray.html
got introduced in https://github.com/rerun-io/rerun/pull/9109