rerun-io/rerun #9863: Google Colab locks up with callbacks on notebook use

pablovela5620, 2025-05-01:
**Describe the bug**

Colab doesn't accurately send callback values when using the notebook package. I've included a video to better demonstrate. I'm running essentially the same code on colab + my local machine. On my local machine (excluding AnnotationContext error described #9862 ) things work fine

https://github.com/user-attachments/assets/41259be1-9d2b-4aeb-9732-d0e3579a41eb

**To Reproduce**
Steps to reproduce the behavior:
1. go to https://colab.research.google.com/drive/140m3EVzX5DScZQOMS5wt0dawtpUecS6m?usp=sharing
2. Run code all the way to the end
3. Use the UI sliders

**Expected behavior**

Callbacks should work the same both on local + colab, local works, colab does not

**Backtrace**

**Desktop (please complete the following information):**
 - OS: 
Local machine is running
(default) pablo@pablo-dl-server:~/0Dev/repos/pi0-lerobot$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.2 LTS
Release:        24.04
Codename:       noble

**Rerun version**

```
rerun-cli 0.23.1 (default map_view nasm native_viewer release web_viewer) [rustc 1.84.0 (9fc6b4312 2025-01-07), LLVM 19.1.5] x86_64-unknown-linux-gnu release-0.23.1 87bdacb, built 2025-04-25T11:04:25Z
Video features: av1 default ffmpeg nasm serde
```
**Additional context**

abey79, 2025-05-05:
FWIW, here are the dependencies that are being installed. (I checked to verify that this wasn't another numpy<2 issue.)

```
absl-py==1.4.0
accelerate==1.6.0
aiohappyeyeballs==2.6.1
aiohttp==3.11.15
aiosignal==1.3.2
alabaster==1.0.0
albucore==0.0.24
albumentations==2.0.6
ale-py==0.11.0
altair==5.5.0
annotated-types==0.7.0
anyio==4.9.0
anywidget==0.9.18
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
array_record==0.7.2
arviz==0.21.0
astropy==7.0.1
astropy-iers-data==0.2025.4.28.0.37.27
astunparse==1.6.3
atpublic==5.1
attrs==25.3.0
audioread==3.0.1
autograd==1.7.0
babel==2.17.0
backcall==0.2.0
backports.tarfile==1.2.0
beartype==0.20.2
beautifulsoup4==4.13.4
betterproto==2.0.0b6
bigframes==2.1.0
bigquery-magics==0.9.0
bleach==6.2.0
blinker==1.9.0
blis==1.3.0
blosc2==3.3.1
bokeh==3.7.2
Bottleneck==1.4.2
bqplot==0.12.44
branca==0.8.1
build==1.2.2.post1
CacheControl==0.14.2
cachetools==5.5.2
catalogue==2.0.10
certifi==2025.4.26
cffi==1.17.1
chardet==5.2.0
charset-normalizer==3.4.1
chex==0.1.89
clarabel==0.10.0
click==8.1.8
cloudpathlib==0.21.0
cloudpickle==3.1.1
cmake==3.31.6
cmdstanpy==1.2.5
colorcet==3.1.0
colorlover==0.3.0
colour==0.1.5
community==1.0.0b1
confection==0.1.5
cons==0.4.6
contourpy==1.3.2
cramjam==2.10.0
cryptography==43.0.3
cuda-python==12.6.2.post1
cudf-cu12 @ https://pypi.nvidia.com/cudf-cu12/cudf_cu12-25.2.1-cp311-cp311-manylinux_2_24_x86_64.manylinux_2_28_x86_64.whl
cudf-polars-cu12==25.2.2
[... 526 lines cut ...]
wasabi==1.1.3
wcwidth==0.2.13
weasel==0.4.1
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
websockets==15.0.1
Werkzeug==3.1.3
widgetsnbextension==3.6.10
wordcloud==1.9.4
wrapt==1.17.2
wurlitzer==3.1.1
xarray==2025.3.1
xarray-einstats==0.8.0
xgboost==2.1.4
xlrd==2.0.1
xyzservices==2025.4.0
yarl==1.20.0
ydf==0.11.0
yellowbrick==1.5
yfinance==0.2.57
zict==3.0.0
zipp==3.21.0
zstandard==0.23.0
```

abey79, 2025-05-05:
Also, I "works" for me, both in safari and chrome, albeit erratically and with a lot of delay (so basically unusable).

pablovela5620, 2025-05-06:
> Also, I "works" for me, both in safari and chrome, albeit erratically and with a lot of delay (so basically unusable).

Yea that was also my experience, since I used the depts already included in colab the numpy 2 issue doesn't occur

Wumpf, 2025-06-29:
@pablovela5620 @jprochazk did we ever find out more about this? What's the action item here for us going from here? 🤔

ntjohnson1, 2025-07-15:
Note that the newer beartype is just a bug in ipython_beartype which should get resolved in the next release https://github.com/beartype/ipython-beartype/issues/6