rerun-io/rerun #9898: rerun_cpp_sdk does not compile on windows

alokpr, 2025-05-07:
Multiple compiler errors like:
```
C:\re_cwd\third-party\rerun_cpp_sdk\src\rerun\loggable.hpp(18): error C2338: static_assert failed: 'Loggable is not implemented for this type. It is implemented for all built-in datatypes and components. To check ahead of template instantiation whether a type is loggable, use `is_loggable<T>`'
C:\re_cwd\third-party\rerun_cpp_sdk\src\rerun\loggable.hpp(18): note: the template instantiation context (the oldest one first) is
C:\re_cwd\third-party\rerun_cpp_sdk\src\rerun\archetypes\../component_batch.hpp(68): note: see reference to class template instantiation 'rerun::Loggable<T>' being compiled
        with
        [
            T=rerun::Collection<rerun::components::Blob>
        ]
```

It seems like function overloading for collection and component is not working as expected on windows:
```
template <typename T>
static Result<ComponentBatch> from_loggable(
            const rerun::Collection<T>& components,
            const ComponentDescriptor& descriptor = rerun::Loggable<T>::Descriptor
) 

template <typename T>
static Result<ComponentBatch> from_loggable(
            const T& component,
            const ComponentDescriptor& descriptor = rerun::Loggable<T>::Descriptor
) 
```

alokpr, 2025-05-07:
@Wumpf Do you see it on your end or is this something specific to the compiler we are using?

Wumpf, 2025-05-07:
@alokpr what compiler are you using? We test on CI with VS2022 (`14.43.34808` to be accurate https://github.com/rerun-io/rerun/actions/runs/14876516385/job/41774776838)

alokpr, 2025-05-07:
We are using VS2022 `14.39.33519` with c++20.

It looks like the compiler does not treat `from_loggable(const rerun::Collection<T>&)` a more specialized overload of `from_loggable(const T&)` and is matching the call for collection-types to `from_loggable(const T&)`. Might need to make the substitution less ambiguous.

Wumpf, 2025-05-08:
Strange, seems unlikely that the minor version difference would cause this. More likely none of our examples & tests are triggering this error.
It's not pretty but I figure we could make the resolution less ambiguous by using SFINAE to disable overloads that aren't valid (i.e. disable `from_loggable(const T&)` iff `T` is a `rerun::Collection`) 🤔 

Do you get this error on the most minimal rerun examples or is there specific calls you have to do to run into this?

alokpr, 2025-05-08:
It might have to do with C++20. Would you mind enabling C++20 if possible?

Wumpf, 2025-05-10:
Was able to repro the issue with C++20 enabled on Windows 👍 . We'll have to add this to our test matrix

Wumpf, 2025-05-19:
We'll do a 0.23.3 release soon which will include the fix

closed 2025-05-19 by Wumpf