rerun-io/rerun #9909: RecordingStream with a pending connection hangs forever

alokpr, 2025-05-07:
Run this snippet **without** any instance of rerun:
```
fn main() {
  let _stream = RecordingStreamBuilder::new("myapp").connect_tcp().unwrap();
}
```

RecordingStream will keep trying to connect in the background and `drop(_stream)` will hang:
```
2025-05-07T23:12:38.598404Z DEBUG re_sdk_comms::buffered_client: Connecting to remote 127.0.0.1:9876…
2025-05-07T23:12:38.598590Z DEBUG re_chunk::batcher: creating new chunk batcher config=ChunkBatcherConfig { flush_tick: 8ms, flush_num_bytes: 1048576, flush_num_rows: 18446744073709551615, chunk_max_rows_if_unsorted: 256, max_commands_in_flight: None, max_chunks_in_flight: None, hooks: BatcherHooks { on_insert: None, on_release: None } }
2025-05-07T23:12:38.598607Z DEBUG re_sdk::recording_stream: setting recording info app_id=myapp rec_id=2d12a7ce-3fc5-4569-8909-2b5dc07c5bfb

2025-05-07T23:12:38.598839Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:38.598967Z DEBUG re_sdk_comms::buffered_client: Failed to send message: Failed to connect to Rerun server at 127.0.0.1:9876: Connection refused (os error 111)
2025-05-07T23:12:38.699066Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:38.899274Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:38.899378Z  WARN re_sdk_comms::buffered_client: Failed to send message after 3 attempts: Failed to connect to Rerun server at 127.0.0.1:9876: Connection refused (os error 111)
2025-05-07T23:12:39.299477Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:40.099659Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:41.699854Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…

CTRL+C

2025-05-07T23:12:42.526979Z DEBUG re_sdk_comms::buffered_client: Shutting down the client connection…
2025-05-07T23:12:42.527017Z DEBUG re_sdk_comms::buffered_client: Flushing message queue…
2025-05-07T23:12:44.700064Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:47.700264Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:50.700464Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
2025-05-07T23:12:53.700654Z DEBUG re_sdk_comms::tcp_client: Connecting to 127.0.0.1:9876…
```

alokpr, 2025-05-07:
There is one workaround - call RecordingStream::disconnect explicitly

Wumpf, 2025-05-08:
@alokpr any chance you could try with a newer version of Rerun? We've been changing the connection stack quite a bit in 0.23 so this may or may not already be fixed. (it's `connect_grpc` on latest)

alokpr, 2025-05-08:
Yes, will do - as soon as I am able to resolve https://github.com/rerun-io/rerun/issues/9898, which is blocking us from v0.22 -> v0.23 upgrade.

jprochazk, 2025-05-23:
The expected behavior on `0.23` and above is that the application shuts down due to the default flush timeout on disconnect. That flush timeout is configurable, and can even be omitted so that it keeps attempting to reconnect until you terminate it manually.

Testing this on `0.23.2`:

```
cargo new repro
cd repro && cargo add rerun
cat <<EOF > src/main.rs
use rerun::RecordingStreamBuilder;
use rerun::external::re_log;
fn main() {
    re_log::setup_logging();
    let _stream = RecordingStreamBuilder::new("myapp").connect_grpc().unwrap();
}
EOF
cargo run
```

I get the following logs:

<details><summary>Logs</summary>

```
[2025-05-23T10:45:44Z DEBUG re_chunk::batcher] creating new chunk batcher config=ChunkBatcherConfig { flush_tick: 8ms, flush_num_bytes: 1048576, flush_num_rows: 18446744073709551615, chunk_max_rows_if_unsorted: 256, max_commands_in_flight: None, max_chunks_in_flight: None, hooks: BatcherHooks { on_insert: None, on_release: None } }
[2025-05-23T10:45:44Z DEBUG re_sdk::recording_stream] setting recording info app_id=myapp rec_id=17c83d20a09a46039ac8e2ebe82dcad4
[2025-05-23T10:45:44Z DEBUG re_sdk::recording_stream] adding recording properties to batcher properties=RecordingProperties { start_time: Some(SerializedComponentBatch { array: PrimitiveArray<Int64>
    [
      1747997144306225131,
    ], descriptor: ComponentDescriptor { archetype_name: Some("rerun.archetypes.RecordingProperties"), archetype_field_name: Some("start_time"), component_name: "rerun.components.Timestamp" } }), name: None }
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] Shutting down message proxy client
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:44Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:45Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:46Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:47Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:47Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:47Z DEBUG re_grpc_client::message_proxy::write] failed to connect to message proxy server: transport error
[2025-05-23T10:45:47Z WARN  re_grpc_client::message_proxy::write] Flush timed out, not all messages were sent. The timeout can be adjusted when connecting via gRPC.
[2025-05-23T10:45:47Z DEBUG re_grpc_client::message_proxy::write] shutting down client without flush
[2025-05-23T10:45:47Z DEBUG re_grpc_client::message_proxy::write] Message proxy client has shut down
```

</details>

And the program exits within 3 seconds. That seems like the expected behavior

jprochazk, 2025-05-23:
Leaving this open for now, as we'd still like to get confirmation that it also works for you once you upgrade :)

Wumpf, 2025-07-16:
thanks for following up!