rerun-io/rerun #9926: Copy Paste From Timeline Timestamp on Mac Copies Recording UUID

ntjohnson1, 2025-05-08:
**Describe the bug**
On Mac when I cmd+c on the timeline timestamp it copies the uuid for the recording instead. If I right click I can copy the timestamp.
Video example:
https://drive.google.com/file/d/1T5yBJ82-tVHaooakOhoOrPlCLKAccIIJ/view?usp=drive_link
Looks related but not exactly the same as: https://github.com/rerun-io/rerun/issues/8338

**To Reproduce**
Steps to reproduce the behavior:
1. Launch a rerun recording. Try to copy and paste the timestamp with keyboard shortcuts.

**Expected behavior**
Highlighted text gets copied.

**Screenshots**
See video

**Desktop (please complete the following information):**
 - OS: macOS Sequoia 15.3.1

**Rerun version**
rerun-cli 0.23.1 (default map_view nasm native_viewer release web_viewer) [rustc 1.84.0 (9fc6b4312 2025-01-07), LLVM 19.1.5] aarch64-apple-darwin release-0.23.1 87bdacb, built 2025-04-25T11:19:56Z
Video features: av1 default ffmpeg nasm serde

Wumpf, 2025-05-08:
Looks like this logic here is incorrectly tripping off here
https://github.com/rerun-io/rerun/blob/40c1c720d86e262f76b02d0c013d3e2c1eab88a5/crates/viewer/re_viewer/src/app_state.rs#L683

jprochazk, 2025-05-23:
Doesn't seem Mac-specific, can repro on Linux

closed 2025-05-23 by Wumpf