rerun-io/rerun #9941: Draw order of bounding boxes in detect & track object example is wrong

Wumpf, 2025-05-09:
https://rerun.io/viewer?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.23.2%2Fexamples%2Fdetect_and_track_objects.rrd

Some of the boxes don't show when they should!

[image]

All boxes have draw order `10.0`. Changing it on the ones that don't show makes them show up, strange!

Wumpf, 2025-05-09:
Seems to happen only on Web: repro'd with prebuilt main on web but not on native. But curiously it's not a webgl only issue (which often has depth issues) 🤔

Wumpf, 2025-05-09:
Also got hold of it on native now, it's juset more fleeting!

pablovela5620, 2025-05-09:
I think I'm having a similar issue. Here bounding boxes show up find in the 3d spatial view, but not in the 2d views

https://app.rerun.io/version/0.23.2/index.html?url=https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/self-collected-wb-top%2Brbl.rrd

[image]

Wumpf, 2025-05-09:
Seems to be specific for videos. Should add a draw order to those

Wumpf, 2025-05-09:
* Duplicate of https://github.com/rerun-io/rerun/issues/9138

closed 2025-05-09 by Wumpf