rerun-io/rerun #9948: Initializing twice with same recording id hangs the SDK

Wumpf, 2025-05-09:
Repeated `init`calls will cause the SDK to hang on exit:
```py
import rerun as rr

rr.init("app1", spawn=True)
rr.log("point", rr.Points3D([1, 2, 3], colors=[255, 0, 0]))

rr.init("app1", spawn=True)
rr.log("point", rr.Points3D([1, 2, 3], colors=[0, 255, 0]))
```

Note that as per documentation this should log to the same recording stream since the recording id is generated once. This works as documented with explicit recording streams:

```py
import rerun as rr

with rr.RecordingStream("app1") as rec:
    rec.spawn()
    rec.log("point", rr.Points3D([1, 2, 3], colors=[255, 0, 0]))

with rr.RecordingStream("app1") as rec:
    rec.spawn()
    rec.log("point", rr.Points3D([1, 2, 3], colors=[0, 255, 0]))
```

Curiously, when using `rr.init` more "realistically" with different recording ids, we don't get a hang on shutdown:
```py
from uuid import uuid4
import rerun as rr

rr.init("app1", recording_id=uuid4(), spawn=True)
rr.log("point", rr.Points3D([1, 2, 3], colors=[255, 0, 0]))

rr.init("app1", recording_id=uuid4(), spawn=True)
rr.log("point", rr.Points3D([1, 2, 3], colors=[0, 255, 0]))
```

Tested on 0.23.2

leonmkim, 2025-06-01:
I have a use case where I explicitly want to save different `.rrd` files with the same recording id so that I can merge the data together from different files after the fact. 

E.g. something like:

```
import rerun as rr
import uuid

recording_id = str(uuid.uuid4())

rr.init("app1", recording_id=recording_id, spawn=False)
rr.save("recording_0.rrd")
rr.log("point", rr.Points3D([1, 2, 3], colors=[255, 0, 0]))

# Will hang here
rr.init("app1", recording_id=recording_id, spawn=False)
rr.save("recording_1.rrd")
rr.log("point", rr.Points3D([1, 2, 3], colors=[0, 255, 0]))
```

However, I ran into this bug which would cause the code to hang when the second init was called. 

That said, it seems adding an `rr.disconnect()` in between `rr.init()` solves the issue.

```
import rerun as rr
import uuid

recording_id = str(uuid.uuid4())

rr.init("app1", recording_id=recording_id, spawn=False)
rr.save("recording_0.rrd")
rr.log("point", rr.Points3D([1, 2, 3], colors=[255, 0, 0]))
rr.disconnect()

# Code executes properly!
rr.init("app1", recording_id=recording_id, spawn=False)
rr.save("recording_1.rrd")
rr.log("point", rr.Points3D([1, 2, 3], colors=[0, 255, 0]))
```

AhmedMousa-ag, 2025-06-08:
There's a solution for it, not sure if it suits your architecture as I'm not fully aware of the inner working of rerun but this is my attempt.

In the rust code, the issue happens when it tries to insert a new recording with the same id into the recordings hashmaps. Not sure if it's a deadlock, I checked for deadlocks using parking_lot but found none, however the behavior seems as a deadlock and couldn't identify yet what is exactly the issue.
It only happens if you attempt to insert a new recording after the first initalization.

This code fixes it in this file: [`rerun_py/src/python_bridge.rs`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/python_bridge.rs#L268), specifically within the `new_recording` function on line 268.

```rust
// NOTE: The Rust-side of the bindings must be in control of the lifetimes of the recordings!
    let mut all_recs = all_recordings();
    // Check if recording already exists first.
    // TODO move it up at the start of the function.
    if let Some(existing_recording) = all_recs.get(&recording_id) {
        return Ok(PyRecordingStream(existing_recording.clone()));
    }
    all_recs.insert(recording_id, recording.clone());
```

I assume avoding insertion of the same recording id could be a good solution as it's mentioned in your documentation that creation of two recordings with the same id is useless and will only create one.

Does this help?

AhmedMousa-ag, 2025-06-09:
My pleasure, I will check your guidelines, code and proceed forward with the unit test.

> [@AhmedMousa-ag](https://github.com/AhmedMousa-ag) thanks for checking on that! Yeah sounds like a viable solution. I think we should also emit a warning when this happens and explain what happens because it's most likely not intentional. Would you up for creating PR with that + a short unit test? :)

PR #10201 (2025-06-15): Fix initializing two recordings with the same recording id causing SDK hangs
### Related

* Closes #9948

### What

* The hang in the issue #9948 is caused when trying to insert duplicated recording, and since we don't need this behavior, just avoid inserting the new recording. The real reason for hang not known yet, but it's not replicated in PyTests.

* Moved the py_rerun_warn function from `rerun_py/src/dataframe.rs` to `rerun_py/src/utils.rs`  for readability.

* Added a new function that does both logging the warning using `re_log::warn` macro and `PyErr::warn` to detect the warning in py [... cut]