rerun-io/rerun #9949: Support to mcap files

patrickelectric, 2025-05-09:
**Is your feature request related to a problem? Please describe.**
When I run a mcap file it says:
```
rerun /tmp/demo_2025-05-08_17-58-25.mcap               
[2025-05-09T18:14:55Z INFO  re_data_loader::load_file] Loading "/tmp/demo_2025-05-08_17-58-25.mcap"…
Error: "/tmp/demo_2025-05-08_17-58-25.mcap" -> No data-loader support for "/tmp/demo_2025-05-08_17-58-25.mcap"
```

**Describe the solution you'd like**
Be able to run mcap files

**Describe alternatives you've considered**
Not that I'm aware

**Additional context**
Just recorded a file in https://mcap.dev/ and rerun failed to play

patrickelectric, 2025-05-13:
references: https://rerun.io/blog/rosbag

Wumpf, 2025-05-13:
Can't believe we didn't have an issue for this already! This is really high on our priority list. Stay tuned!

patrickelectric, 2025-06-04:
Just for curiosity.. and my sanity.. is there any timeline for this or prediction ? 
We are working to integrate zenoh + mcap into our system and would like to use rerun

patrickelectric, 2025-06-09:
Thanks @nikolausWest! I'll certainly help with that

nikolausWest, 2025-06-09:
@patrickelectric amazing, thanks!

git-pi-e, 2025-06-18:
Foxglove also has a native protobuf support, so MCAP and protobuf together would also be great.

grtlr, 2025-07-30:
> I believe it would be great if alongside mcap, there is a native support for protobuf deserializing too, similar to Foxglove. 

That sounds like something we have planned. Do you mean deserializing arbitrary messages that are encoded with protobuf and ingesting their fields as Rerun [`AnyValues`](https://rerun.io/docs/howto/logging/custom-data)?