rerun-io/rerun #10030: Tagged component Viewer-UI/UX implications

gavrelina, 2025-05-21:
## Context
We introduce tagged components and this shall be reflected throughout UI/UX of our Viewer. 
Before: one entity path could have had a several `Color` components
Now: thanks to tagged components we are now aware of which archetype each of these `Color` components belongs to. Also we have more semantic names instead of "data type" names. 

## Designs
Approach: group by archetypes. Use very similar styling for "marking" archetypes, in thie examples below its always a bit grey and smaller font, as it's more of "an indicator" rather than the data itself. 
[Figma section](https://www.figma.com/design/72AuvfbssFJeqqnQoJscYT/Katya-s-playground?node-id=5723-4487&t=Yt0vSGvMfcjfQlEo-1) **_with more detailed instructions_**

example of spec:
[image]

### Streams panel
[image]

### Selection panel
[image]. [image]

### Dataframe view
[image]

**Additional context**
Related to https://github.com/rerun-io/rerun/issues/6889
Related to https://github.com/rerun-io/rerun/issues/8368
Related to https://github.com/rerun-io/rerun/issues/3231

nikolausWest, 2025-05-21:
Very nice in general. Nit-y question: in the panels you write out "Pinhole Archetype" but in the Dataframe View you just write "Pinhole". Is there any thinking behind the difference? My gut feeling is I don't like exposing the Archetype word so prominently since it isn't really a concept you have to understand until you start going deeper but this is a very blurry line and I haven't thought deeply about this

gavrelina, 2025-05-21:
@nikolausWest wanted to add `archetype` to the selection panel because we also display visualizers in next to components. Sometimes the names of visualizers and archetypes are the same. So explicitly adding `Mesh3D visualizer` and `Mesh3D archetype` could help users understand a difference. I would also add that to Streams panel, as it's a similar way of listing components.  
However in a table the context is different (mostly visually):
- components are distributed horizontally and not placed all under 1 archetype type, so repeating `archetype` every time for each cell will look too repetitive
- cell is smaller and adding archetype will also increase the length. plus if it's longer than the the field name it will look a bit unbalanced. 
- also the new tooltip for each column details that Antoine was introducing, can really give more info there!
So as a summary of above I decided to spell out archetype in 2 places and not in 3 :)

nikolausWest, 2025-05-22:
Makes sense. What do you think of hiding the archetype / visualizer copy in the panels behind tooltips as well.

There is a risk we need to think about where users find Rerun intimidating with new concepts too early

PR #10149 (2025-06-11): Group dataframe table by archetype and use new table design
### Related

* closes #10030 

### What

Updates the dataframe table for the new table design and adds the archetype name to the component headers.

The column dividers are now rendered by the rerun code, to allow connecting cells visually.

Also fixed the class id color rects being too large.

[image]