rerun-io/rerun #10037: Default to light mode in Jupyter notebooks

emilk, 2025-05-21:
Jupyter notebooks always have a light background. I therefor want Rerun to default to light mode there (and in other “bright” notebook surroundings).

But I otherwise want Rerun to respect the system preferences by default (both on web and on native).

Wumpf, 2025-06-13:
@emilk I saw you and @jprochazk discussing this on Slack. Was there any conclusion on whether this can be done?

jprochazk, 2025-06-13:
We don't yet know how to do this reliably. We can treat it as another "UI property" like panel states, and expose an option in the `Viewer` constructor for it.

But also light mode isn't done yet:
* https://github.com/rerun-io/rerun/pull/10107

emilk, 2025-07-02:
I got it to work:

[image]

PR coming later

Wumpf, 2025-07-14:
Some unclarity whether we actually already fixed this 😳 . Have to check!