rerun-io/rerun #10169: Support non-uniform scaling of X/Y axis in 2D spatial view

emilk, 2025-06-08:
* Part of https://github.com/rerun-io/rerun/issues/1630

This unlocks the ability to use the 2D view as a generic plot area, using `Points2D` for scatter-plots and `LineStrips2D` for lines.

nikolausWest, 2025-06-08:
I’ve been thinking a bit lately about us pushing for using the spatial 2D view for plots and although I was originally probably the strongest proponent, I’m not so sure anymore.

We’d like to move to a world where logging is about modeling the semantics of the data and how that then gets shown is separate (declared in blueprint). If we exposed a really powerful data transformation and remapping system it wouldn’t matter, but we currently don’t and most likely that system will start simple when we do expose it. I think it will be much easier to build a system that picks data from anywhere and sets to X ir Y series of a plot, than transforming to spatial archetypes. That makes me favor trying to add non-time axes to the plot view instead

mightymichael44, 2026-02-18:
Hey Folks,

is there by any chance an update on the topic?
I am looking at rerun at it looks like a perfect fit however the limited 2D plotting makes me hesitate.