rerun-io/rerun #10190: Error when trying to build the Python SDK from source (MacOS)

pweids, 2025-06-10:
From a relatively clean MacOS install (Python v3.13.4, rust version v1.85, pixi v0.48.0, Node v24.2.0), I followed the instructions in BUILD.md

```
git lfs install
pixi run check-env
pixi run rerun # worked fine
pixi run py-build --release
```

The error I get is:

```
% pixi run py-build --release
✨ Pixi task (py-build in py): pixi run -e py py-build-common --release
Error:   × Failed to update PyPI packages for environment 'py'
  ├─▶ Failed to prepare distributions
  ├─▶ Failed to build `rerun-notebook @ file:///Users/paul/Development/rerun/rerun_notebook`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `hatchling.build.build_editable` failed (exit status: 1)

      [stdout]

      up to date, audited 87 packages in 527ms

      10 packages are looking for funding
        run `npm fund` for details

      3 vulnerabilities (2 moderate, 1 high)

      To address issues that do not require attention, run:
        npm audit fix

      To address all issues (including breaking changes), run:
        npm audit fix --force

      Run `npm audit` for details.

      > build
      > node build.mjs

      [stderr]
      INFO:hatch_jupyter_builder.utils:Running jupyter-builder
      INFO:hatch_jupyter_builder.utils:Building with hatch_jupyter_builder.npm_builder
      INFO:hatch_jupyter_builder.utils:With kwargs: {'npm': 'npm', 'build_cmd': 'build'}
      INFO:hatch_jupyter_builder.utils:Installing build dependencies with npm.  This may take a while...
      INFO:hatch_jupyter_builder.utils:> /Users/paul/Development/rerun/.pixi/envs/py/bin/npm install
      INFO:hatch_jupyter_builder.utils:> /Users/paul/Development/rerun/.pixi/envs/py/bin/npm run build
      ✘ [ERROR] Could not resolve "@rerun-io/web-viewer/inlined.js"

          src/js/widget.ts:7:7:
            7 │ } from "@rerun-io/web-viewer/inlined.js";
              ╵        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

        The module "./inlined.js" was not found on the file system:

          node_modules/@rerun-io/web-viewer/package.json:45:16:
            45 │       "import": "./inlined.js"
               ╵                 ~~~~~~~~~~~~~~

        You can mark the path "@rerun-io/web-viewer/inlined.js" as external to exclude it from the bundle, which will remove this error and leave the unresolved path in the bundle.

      file:///Users/paul/Development/rerun/rerun_notebook/build.mjs:51
          throw new Error(`Failed to rebuild widget:\n${e.toString()}`);
                ^

      Error: Failed to rebuild widget:
      Error: Build failed with 1 error:
      src/js/widget.ts:7:7: ERROR: Could not resolve "@rerun-io/web-viewer/inlined.js"
          at build (file:///Users/paul/Development/rerun/rerun_notebook/build.mjs:51:11)
          at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
          at async file:///Users/paul/Development/rerun/rerun_notebook/build.mjs:13:3

      Node.js v20.17.0
      Traceback (most recent call last):
        File "<string>", line 11, in <module>
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpFQplYl/lib/python3.11/site-packages/hatchling/build.py", line 83, in build_editable
          return os.path.basename(next(builder.build(directory=wheel_directory, versions=['editable'])))
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpFQplYl/lib/python3.11/site-packages/hatchling/builders/plugin/interface.py", line 147, in build
          build_hook.initialize(version, build_data)
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpFQplYl/lib/python3.11/site-packages/hatch_jupyter_builder/plugin.py", line 94, in initialize
          raise e
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpFQplYl/lib/python3.11/site-packages/hatch_jupyter_builder/plugin.py", line 89, in initialize
          build_func(self.target_name, version, **build_kwargs)
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpFQplYl/lib/python3.11/site-packages/hatch_jupyter_builder/utils.py", line 117, in npm_builder
          run([*npm_cmd, "run", build_cmd], cwd=str(abs_path))
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpFQplYl/lib/python3.11/site-packages/hatch_jupyter_builder/utils.py", line 231, in run
          return subprocess.check_call(cmd, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/paul/Development/rerun/.pixi/envs/py/lib/python3.11/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['/Users/paul/Development/rerun/.pixi/envs/py/bin/npm', 'run', 'build']' returned non-zero exit status 1.

      hint: This usually indicates a problem with the package or the build environment.
```

The primary issue being:
```
✘ [ERROR] Could not resolve "@rerun-io/web-viewer/inlined.js"
```

I'm likely missing a dependency (the web-viewer), but not sure how to install that. It should also be added to the build documentation.

Wumpf, 2025-06-10:
looks like something in `pixi run py-build` is indeed dragging in the web viewer when it shouldn't
running `pixi run rerun-build-web` once is likely fixing this, but takes quite a while which is why we don't want `pixi run py-build` depend on it (`py-build` is meant for fast-ish turnaround when testing things from python)

pweids, 2025-06-11:
I tried `pixi run rerun-build-web`, but unfortunately hit another blocker with LLVM

`% pixi run rerun-build-web
✨ Pixi task (rerun-build-web in default): rustup target add wasm32-unknown-unknown && cargo run --quiet -p re_dev_tools -- build-web-viewer --no-default-features --features analytics,map_view --debug
info: component 'rust-std' for target 'wasm32-unknown-unknown' is up to date
Building web viewer…

Compiling Rust to wasm in /Users/paul/Development/rerun/target_wasm…
/Users/paul/Development/rerun> CARGO_ENCODED_RUSTFLAGS="--cfg=web_sys_unstable_apis" RUSTFLAGS="--cfg=web_sys_unstable_apis" "cargo" "build" "--quiet" "--package=re_viewer" "--lib" "--target=wasm32-unknown-unknown" "--target-dir=/Users/paul/Development/rerun/target_wasm" "--no-default-features" "--features=analytics,map_view"
error: failed to build archive at `/Users/paul/Development/rerun/target_wasm/wasm32-unknown-unknown/debug/deps/libpsm-7e06df504b029e51.rlib`: LLVM error: section too large

error: could not compile `psm` (lib) due to 1 previous error
Error: Failed to build Wasm`

pweids, 2025-06-11:
@Wumpf that workaround did work, but the original problem persists somehow:

```
% pixi run rerun-build-web
✨ Pixi task (rerun-build-web in default): rustup target add wasm32-unknown-unknown && cargo run --quiet -p re_dev_tools -- build-web-viewer --no-default-features --features analytics,map_view --debug
info: component 'rust-std' for target 'wasm32-unknown-unknown' is up to date
Building web viewer…

Compiling Rust to wasm in /Users/paul/Development/rerun/target_wasm…
/Users/paul/Development/rerun> CARGO_ENCODED_RUSTFLAGS="--cfg=web_sys_unstable_apis" RUSTFLAGS="--cfg=web_sys_unstable_apis" "cargo" "build" "--quiet" "--package=re_viewer" "--lib" "--target=wasm32-unknown-unknown" "--target-dir=/Users/paul/Development/rerun/target_wasm" "--no-default-features" "--features=analytics,map_view"
Web viewer .wasm built in 156.1s

Generating JS bindings for wasm…
Generated JS bindings in 2.5s

Finished /Users/paul/Development/rerun/web_viewer/re_viewer_bg.wasm
```
then...
```
% pixi run py-build --release
✨ Pixi task (py-build in py): pixi run -e py py-build-common --release
Error:   × Failed to update PyPI packages for environment 'py'
  ├─▶ Failed to prepare distributions
  ├─▶ Failed to build `rerun-notebook @ file:///Users/paul/Development/rerun/rerun_notebook`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `hatchling.build.build_editable` failed (exit status: 1)

      [stdout]

      up to date, audited 87 packages in 883ms

      10 packages are looking for funding
        run `npm fund` for details

      3 vulnerabilities (2 moderate, 1 high)

      To address issues that do not require attention, run:
        npm audit fix

      To address all issues (including breaking changes), run:
        npm audit fix --force

      Run `npm audit` for details.

      > build
      > node build.mjs

      [stderr]
      INFO:hatch_jupyter_builder.utils:Running jupyter-builder
      INFO:hatch_jupyter_builder.utils:Building with hatch_jupyter_builder.npm_builder
      INFO:hatch_jupyter_builder.utils:With kwargs: {'npm': 'npm', 'build_cmd': 'build'}
      INFO:hatch_jupyter_builder.utils:Installing build dependencies with npm.  This may take a while...
      INFO:hatch_jupyter_builder.utils:> /Users/paul/Development/rerun/.pixi/envs/py/bin/npm install
      INFO:hatch_jupyter_builder.utils:> /Users/paul/Development/rerun/.pixi/envs/py/bin/npm run build
      ✘ [ERROR] Could not resolve "@rerun-io/web-viewer/inlined.js"

          src/js/widget.ts:7:7:
            7 │ } from "@rerun-io/web-viewer/inlined.js";
              ╵        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

        The module "./inlined.js" was not found on the file system:

          node_modules/@rerun-io/web-viewer/package.json:45:16:
            45 │       "import": "./inlined.js"
               ╵                 ~~~~~~~~~~~~~~

        You can mark the path "@rerun-io/web-viewer/inlined.js" as external to exclude it from the bundle, which will remove this error and leave the unresolved path in the bundle.

      file:///Users/paul/Development/rerun/rerun_notebook/build.mjs:51
          throw new Error(`Failed to rebuild widget:\n${e.toString()}`);
                ^

      Error: Failed to rebuild widget:
      Error: Build failed with 1 error:
      src/js/widget.ts:7:7: ERROR: Could not resolve "@rerun-io/web-viewer/inlined.js"
          at build (file:///Users/paul/Development/rerun/rerun_notebook/build.mjs:51:11)
          at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
          at async file:///Users/paul/Development/rerun/rerun_notebook/build.mjs:13:3

      Node.js v20.17.0
      Traceback (most recent call last):
        File "<string>", line 11, in <module>
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpANkEAe/lib/python3.11/site-packages/hatchling/build.py", line 83, in build_editable
          return os.path.basename(next(builder.build(directory=wheel_directory, versions=['editable'])))
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpANkEAe/lib/python3.11/site-packages/hatchling/builders/plugin/interface.py", line 147, in build
          build_hook.initialize(version, build_data)
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpANkEAe/lib/python3.11/site-packages/hatch_jupyter_builder/plugin.py", line 94, in initialize
          raise e
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpANkEAe/lib/python3.11/site-packages/hatch_jupyter_builder/plugin.py", line 89, in initialize
          build_func(self.target_name, version, **build_kwargs)
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpANkEAe/lib/python3.11/site-packages/hatch_jupyter_builder/utils.py", line 117, in npm_builder
          run([*npm_cmd, "run", build_cmd], cwd=str(abs_path))
        File "/Users/paul/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpANkEAe/lib/python3.11/site-packages/hatch_jupyter_builder/utils.py", line 231, in run
          return subprocess.check_call(cmd, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/paul/Development/rerun/.pixi/envs/py/lib/python3.11/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['/Users/paul/Development/rerun/.pixi/envs/py/bin/npm', 'run', 'build']' returned non-zero exit status 1.

      hint: This usually indicates a problem with the package or the build environment.
```

Wumpf, 2025-06-11:
@abey79 any of this familiar to you?

jccampagne, 2025-06-24:
I have a slightly different outcome than https://github.com/rerun-io/rerun/issues/10190#issuecomment-2962816996

I am running `pixi 0.48.2` on macOS M1 chip.

# pixi run rerun-build-web

`pixi run rerun-build-web` does take a while but it runs OK.

```
% pixi run rerun-build-web
✨ Pixi task (rerun-build-web in default): rustup target add wasm32-unknown-unknown && cargo run --quiet -p re_dev_tools -- build-web-viewer --no-default-features --features analytics,map_view --debug
info: component 'rust-std' for target 'wasm32-unknown-unknown' is up to date
Building web viewer…

Compiling Rust to wasm in /Users/jc/src/rerun-py-debug/target_wasm…
/Users/jc/src/rerun-py-debug> CARGO_ENCODED_RUSTFLAGS="--cfg=web_sys_unstable_apis" RUSTFLAGS="--cfg=web_sys_unstable_apis" "cargo" "build" "--quiet" "--package=re_viewer" "--lib" "--target=wasm32-unknown-unknown" "--target-dir=/Users/jc/src/rerun-py-debug/target_wasm" "--no-default-features" "--features=analytics,map_view"
Web viewer .wasm built in 297.5s

Generating JS bindings for wasm…
Generated JS bindings in 3.5s

Finished /Users/jc/src/rerun-py-debug/web_viewer/re_viewer_bg.wasm
```

However the next step (`pixi run py-build --release`) fails:

# pixi run py-build --release

```
% pixi run py-build --release
✨ Pixi task (py-build in py): pixi run -e py py-build-common --release
Error:   × Failed to update PyPI packages for environment 'py'
  ├─▶ Failed to prepare distributions
  ├─▶ Failed to build `rerun-notebook @ file:///Users/jc/src/rerun-py-debug/rerun_notebook`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `hatchling.build.build_editable` failed (exit status: 1)
      
      [stdout]
      
      added 72 packages, and audited 74 packages in 3s
      
      2 vulnerabilities (1 moderate, 1 high)
      
      To address issues that do not require attention, run:
        npm audit fix
      
      To address all issues (including breaking changes), run:
        npm audit fix --force
      
      Run `npm audit` for details.
      
      > build
      > node build.mjs
      
      
      [stderr]
      INFO:hatch_jupyter_builder.utils:Running jupyter-builder
      INFO:hatch_jupyter_builder.utils:Building with hatch_jupyter_builder.npm_builder
      INFO:hatch_jupyter_builder.utils:With kwargs: {'npm': 'npm', 'build_cmd': 'build'}
      INFO:hatch_jupyter_builder.utils:Installing build dependencies with npm.  This may take a while...
      INFO:hatch_jupyter_builder.utils:> /Users/jc/src/rerun-py-debug/.pixi/envs/py/bin/npm install
      INFO:hatch_jupyter_builder.utils:> /Users/jc/src/rerun-py-debug/.pixi/envs/py/bin/npm run build
      ✘ [ERROR] Could not resolve "./re_viewer"
      
          ../rerun_js/web-viewer/index.ts:7:23:
            7 │   return (await import("./re_viewer")).default;
              ╵                        ~~~~~~~~~~~~~
      
      file:///Users/jc/src/rerun-py-debug/rerun_notebook/build.mjs:26
          throw new Error(`Failed to build widget:\n${e.toString()}`);
                ^
      
      Error: Failed to build widget:
      Error: Build failed with 1 error:
      ../rerun_js/web-viewer/index.ts:7:23: ERROR: Could not resolve "./re_viewer"
          at main (file:///Users/jc/src/rerun-py-debug/rerun_notebook/build.mjs:26:11)
          at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
          at async file:///Users/jc/src/rerun-py-debug/rerun_notebook/build.mjs:44:1
      
      Node.js v20.17.0
      Traceback (most recent call last):
        File "<string>", line 11, in <module>
        File "/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmp3A45c1/lib/python3.11/site-packages/hatchling/build.py", line 83, in build_editable
          return os.path.basename(next(builder.build(directory=wheel_directory, versions=['editable'])))
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmp3A45c1/lib/python3.11/site-packages/hatchling/builders/plugin/interface.py", line 147, in build
          build_hook.initialize(version, build_data)
        File "/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmp3A45c1/lib/python3.11/site-packages/hatch_jupyter_builder/plugin.py", line 94, in initialize
          raise e
        File "/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmp3A45c1/lib/python3.11/site-packages/hatch_jupyter_builder/plugin.py", line 89, in initialize
          build_func(self.target_name, version, **build_kwargs)
        File "/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmp3A45c1/lib/python3.11/site-packages/hatch_jupyter_builder/utils.py", line 117, in npm_builder
          run([*npm_cmd, "run", build_cmd], cwd=str(abs_path))
        File "/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmp3A45c1/lib/python3.11/site-packages/hatch_jupyter_builder/utils.py", line 231, in run
          return subprocess.check_call(cmd, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/jc/src/rerun-py-debug/.pixi/envs/py/lib/python3.11/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['/Users/jc/src/rerun-py-debug/.pixi/envs/py/bin/npm', 'run', 'build']' returned non-zero exit status 1.
      
      hint: This usually indicates a problem with the package or the build environment.
```

jccampagne, 2025-06-24:
I was able to make progress by compiling `rerun-web` and copying the files:

```
cp -R web_viewer/* rerun_js/web-viewer/
```

```
ls -l rerun_js/web-viewer/
total 195168
-rw-r--r--  1 jc  staff      5038 24 Jun 10:19 build-wasm.mjs
-rw-r--r--  1 jc  staff     15086 24 Jun 15:42 favicon.ico
-rw-r--r--  1 jc  staff      1006 24 Jun 15:42 favicon.svg
-rw-r--r--  1 jc  staff     14505 24 Jun 15:42 index.html
-rw-r--r--  1 jc  staff     27334 24 Jun 10:19 index.ts
-rw-r--r--  1 jc  staff       312 24 Jun 15:42 manifest.json
-rw-r--r--  1 jc  staff      1489 24 Jun 10:19 package.json
-rw-r--r--  1 jc  staff  99623631 24 Jun 15:42 re_viewer_bg.wasm
-rw-r--r--  1 jc  staff    201133 24 Jun 15:42 re_viewer.js
-rw-r--r--  1 jc  staff       136 24 Jun 15:42 README.md
-rw-r--r--  1 jc  staff       589 24 Jun 15:42 sw.js
-rw-r--r--  1 jc  staff       297 24 Jun 10:19 tsconfig.json
```

`pixi run py-build` progresses further, but then it fails later on a different issue (that I cam across earlier):

```
% pixi run py-build
✨ Pixi task (py-build in py): pixi run -e py py-build-common
Error:   × Failed to update PyPI packages for environment 'py'
  ├─▶ Failed to prepare distributions
  ├─▶ Failed to build `rerun-sdk @ file:///Users/jc/src/rerun-py-debug/rerun_py`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `maturin.build_editable` failed (exit status: 1)
      
      [stdout]
      Running `maturin pep517 build-wheel -i /Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpuj0YLD/bin/python --compatibility off --editable`
      
      [stderr]
      🍹 Building a mixed python/rust project
      🔗 Found pyo3 bindings with abi3 support
      🐍 Not using a specific python interpreter
      📡 Using build options locked, config from pyproject.toml
         Compiling pyo3-build-config v0.24.2
         Compiling arrow-select v55.1.0

...(snip snip)...

         Compiling re_datafusion v0.24.0-alpha.1+dev (/Users/jc/src/rerun-py-debug/crates/store/re_datafusion)
         Compiling datafusion-ffi v47.0.0
      error: linking with `rust-lld` failed: exit status: 1
        |
        = note: env -u IPHONEOS_DEPLOYMENT_TARGET -u TVOS_DEPLOYMENT_TARGET -u XROS_DEPLOYMENT_TARGET LC_ALL="C" PATH="/Users/jc/.rustup/toolchains/1.85.0-aarch64-apple-darwin/lib/rustlib/aarch64-apple-
      darwin/bin:/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpuj0YLD/bin:/Users/jc/src/rerun-py-debug/.pixi/envs/py/bin:/Users/jc/.cargo/bin:/opt/homebrew/Cellar/llvm/18.1.8/bin/:/

... (snip snip)...

      rerun_bindings.abi3.so" "-undefined" "dynamic_lookup"
        = note: some arguments are omitted. use `--verbose` to show all linker arguments
        = note: rust-lld: error: unknown argument '-Wl,-install_name,@rpath/rerun_bindings.abi3.so'
                
      
      error: could not compile `rerun_py` (lib) due to 1 previous error
      💥 maturin failed
        Caused by: Failed to build a native library through cargo
        Caused by: Cargo build finished with "exit status: 101": `env -u CARGO PYO3_ENVIRONMENT_SIGNATURE="cpython-3.11-64bit" PYO3_PYTHON="/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpuj0YLD/
      bin/python" PYTHON_SYS_EXECUTABLE="/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpuj0YLD/bin/python" "cargo" "rustc" "--message-format" "json-render-diagnostics" "--locked" "--config"
      "env.RERUN_BUILDING_WHEEL=\"1\"" "--manifest-path" "/Users/jc/src/rerun-py-debug/rerun_py/Cargo.toml" "--release" "--lib" "--" "-C" "link-arg=-undefined" "-C" "link-arg=dynamic_lookup" "-C" "link-
      args=-Wl,-install_name,@rpath/rerun_bindings.abi3.so"`
      Error: command ['maturin', 'pep517', 'build-wheel', '-i', '/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmpuj0YLD/bin/python', '--compatibility', 'off', '--editable'] returned non-zero exit
      status 1
      
      hint: This usually indicates a problem with the package or the build environment.

jccampagne, 2025-06-24:
Previous attempts were on master.
I just tried with version `0.23.3`.

The first problem (missing files for js `import`) does not occur; but it fails later on the same linking error:

```
 (git)-[tags/0.23.3]-rerun-prev-version % pixi run py-build                                     
✨ Pixi task (py-build in py): pixi run -e py py-build-common
Error:   × Failed to update PyPI packages for environment 'py'
  ├─▶ Failed to prepare distributions
  ├─▶ Failed to build `rerun-sdk @ file:///Users/jc/src/rerun-prev-version/rerun_py`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `maturin.build_editable` failed (exit status: 1)
      
      [stdout]
      Running `maturin pep517 build-wheel -i /Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmppWvRyJ/bin/python --compatibility off --editable`
      
      [stderr]
          Updating crates.io index
       Downloading crates ...
        Downloaded android_system_properties v0.1.5

...

      blake3-c55815198970233a/out" "-L" "/Users/jc/src/rerun-prev-version/target/release/build/libmimalloc-sys-90a5887cbb43e157/out" "-o" "/Users/jc/src/rerun-prev-
      version/target/release/deps/librerun_bindings.dylib" "-dead_strip" "-dylib" "-undefined" "dynamic_lookup" "-Wl,-install_name,@rpath/rerun_bindings.abi3.so"
      "-undefined" "dynamic_lookup"
        = note: rust-lld: error: unknown argument '-Wl,-install_name,@rpath/rerun_bindings.abi3.so'
                
      
      error: could not compile `rerun_py` (lib) due to 1 previous error
      💥 maturin failed
        Caused by: Failed to build a native library through cargo
        Caused by: Cargo build finished with "exit status: 101": `env -u CARGO PYO3_ENVIRONMENT_SIGNATURE="cpython-3.11-64bit" PYO3_PYTHON="/Users/jc/Library/
      Caches/rattler/cache/uv-cache/builds-v0/.tmppWvRyJ/bin/python" PYTHON_SYS_EXECUTABLE="/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmppWvRyJ/
      bin/python" "cargo" "rustc" "--message-format" "json-render-diagnostics" "--locked" "--config" "env.RERUN_BUILDING_WHEEL=\"1\"" "--manifest-path" "/
      Users/jc/src/rerun-prev-version/rerun_py/Cargo.toml" "--release" "--lib" "--" "-C" "link-arg=-undefined" "-C" "link-arg=dynamic_lookup" "-C" "link-
      args=-Wl,-install_name,@rpath/rerun_bindings.abi3.so"`
      Error: command ['maturin', 'pep517', 'build-wheel', '-i', '/Users/jc/Library/Caches/rattler/cache/uv-cache/builds-v0/.tmppWvRyJ/bin/python',
      '--compatibility', 'off', '--editable'] returned non-zero exit status 1
      
      hint: This usually indicates a problem with the package or the build environment.

```

jccampagne, 2025-06-25:
update: `pixi run py-build` runs fine now for me.
Turns out it was my local rust toolchain.
But I still had to copy the files (`cp -R web_viewer/* rerun_js/web-viewer/`), is that necessary?

Wumpf, 2025-06-25:
> Turns out it was my local rust toolchain.

What did you need to change to fix it? Sounds like we're missing something in the docs there!

> But I still had to copy the files (cp -R web_viewer/* rerun_js/web-viewer/), is that necessary?

that's a weird bug in our build setup that needs fixing. Afaik building the web viewer build should copy things for you, but the fact that that is needed in the first place for py-build is a bug!