rerun-io/rerun #10229: Stop encoding/decoding arrow data in `re_grpc_server`

emilk, 2025-06-13:
`re_grpc_server` is a pure proxy server, who's only job is to relay messages for late-connecting clients. But currently it wastes a lot of CPU time on encoding/decoding arrow-data into protobuff, including expensive IPC encode/decode and LZ4 compress/decompress. As far as I can tell, there is no reason for this.

teh-cmc, 2025-07-08:
> We've talked about this a bit on Slack, and our conclusion is that if we want a smart proxy (and we do, at some point), then we need to at least decode the messages. BUT, we shouldn't need to re-encode them again.

I wasn't part of this discussion so not sure what's involved exactly in "smart proxy" here, but normally all data necessary for routing purposes is present at the transport/protobuf layer: there is never any need to decode application/arrow data during routing (Redap for example never decodes any Arrow).

emilk, 2025-07-08:
Good point