rerun-io/rerun #10280: Can't click top-bar buttons when fullscreen on macOS

emilk, 2025-06-17:
When going fullscreen on macOS, the buttons on the top bar (e.g. panel toggles) _sometimes_ aren't clickable.

Wumpf, 2025-06-19:
Kinda tricky to act upon this without a repro case. Do you have regular access to a setup that has this issue? Otherwise there's nothing we can do here to confirm or deny the existence of this issue in future versions

yash-s20, 2025-06-20:
here's what it looks like on the welcome screen. I hope this helps

https://github.com/user-attachments/assets/815a8305-e014-4a33-a356-46395eec4439

yash-s20, 2025-06-20:
also, this is not run from inside a docker container.

Wumpf, 2025-06-24:
thanks, I was able to repro it now! Kinda strange because I'm always on external monitors, but maybe when I checked last on this I wasn't

Wumpf, 2025-06-24:
curious, `FULLSIZE_CONTENT = false` does still exhibit this behavior!

lucasmerlin, 2025-06-30:
I can reproduce this in the egui demo app, so I don't think this is a rerun only issue. I'll investigate!

lucasmerlin, 2025-06-30:
After a git bisect, this has apparently been broken since/by https://github.com/emilk/egui/pull/2279
But looking at this PR it seems really weird that this change would introduce such bug 😅 Maybe some recent macos update triggers this now?

lucasmerlin, 2025-06-30:
Seems like this is a winit, or even MacOS bug 😬 I've opened a issue there: https://github.com/rust-windowing/winit/issues/4295

Wumpf, 2025-07-14:
Pretty clear that we won't be able to fix this up for 0.24 then. Removing from the milestone