rerun-io/rerun #10353: serve_web function need to back

OpenJarvisAI, 2025-06-25:
The rerun python sdk API changes  tooooo frequent and without further consideration.

First, it's serve, then serve_web, and then deleted serve_web

Please avoid change simple APIs tooo frequently, this caused **all old** (actually not very old) rerun code all breaks~!

def serve_web(*, open_browser=True, web_port=None, ws_port=None, default_blueprint=None, recording=None, server_memory_limit='25%')

Can anybody tells, what's ws_port gone and what's this function replacement with ws_port??

**Don't change APIs toooo frequently, this is why tensorflow dead**

Wumpf, 2025-06-25:
Turns out we actually wanted to deprecate `serve_web` as well for a while now because we found it was easy to get confused by it 😬 

* see https://github.com/rerun-io/rerun/issues/9541

but in line with what I wrote earlier we're keeping it around merely deprecating it.
Also, I added a lot more docs around this stuff now in
* #10360

OpenJarvisAI, 2025-06-26:
Thanks for eaching out and provide so many useful and detail helps.

However, I run into this issue simply by runing Lerobot example. They **still** using very old API:

https://github.com/huggingface/lerobot/blob/fe88c5942cce222c5463350d999a463b9016cf8c/lerobot/scripts/visualize_dataset.py#L57

How to mimic the old behavior in newest rerun sdk?

(They need ws in order to visualize from a remote machine with real-time data playback)

OpenJarvisAI, 2025-06-26:
but lerobot is update sooo slowly, am still don't know how to modify lerobot code to support upcoming rerun 0.24

OpenJarvisAI, 2025-06-27:
thanks, am current use 0.22 forced, however, some of our codes need newest rerun, this caused sort of inconvenient, I hope with 0.24, we can have a minimal code to support both.