rerun-io/rerun #10398: Displaying the Rerun Viewer breaks copying of cell outputs in jupyter

jleibs, 2025-06-26:
Note this ONLY breaks copying from cell outputs, copying from cell inputs for some reason works.

Confirmed as broken on: Linux(Arch) for both Chrome and Firefox

Repro:

* Create a notebook:

Cell 1:

```
import rerun as rr
from rerun.notebook import Viewer
```

Cell 2:

```
print("hello world")
```

Confirm you can copy and paste the output, e.g. just "hello"

Cell 3:

```
rr.init("rerun_example_anything")
rr.notebook_show()
```

Now go back and try to copy and paste, e.g. just "world" — copy buffer is no longer update.

Video demonstration

[1750969945.mp4](https://uploads.linear.app/93919c5e-abe4-4cd5-b77d-7e3eafe3e62d/29d0b5cd-002f-4d1a-9712-46adbe72221d/eb0de2cf-57fd-4ff3-b880-04b8a0204b4a?signature=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJwYXRoIjoiLzkzOTE5YzVlLWFiZTQtNGNkNS1iNzdkLTdlM2VhZmUzZTYyZC8yOWQwYjVjZC0wMDJmLTRkMWEtOTcxMi00NmFkYmU3MjIyMWQvZWIwZGUyY2YtNTdmZC00ZmYzLWI4ODAtMDRiOGEwMjA0YjRhIiwiaWF0IjoxNzUwOTcwMDA3LCJleHAiOjMzMzIxNTMwMDA3fQ.Q7qnJgDxLt20K1KFeJ66bsaFr69urjXHaxA0iLSCI7k)

nikolausWest, 2025-06-26:
Can confirm broken on Mac too

Wumpf, 2025-06-27:
fixed in https://github.com/rerun-io/rerun/pull/10401?