rerun-io/rerun #10447: ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column

fortminors, 2025-06-30:
**Describe the bug**

Minimal Rust demo does not work while building rerun from source - the rerun viewer fails with the following error

```
ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
```

**To Reproduce**
Steps to reproduce the behavior:
1. 
2. 
3. 
4. 

cargo.toml
```toml
[package]
name = "test_rerun"
version = "0.1.0"
edition = "2024"

[dependencies]
rerun = { git = "https://github.com/rerun-io/rerun.git" }
```

main.rs
```rust
use rerun::{demo_util::grid, external::glam};

fn main() -> Result<(), Box<dyn std::error::Error>> {
    let rec = rerun::RecordingStreamBuilder::new("rerun_test").connect_grpc()?;

    let mut counter = 0;

    loop {
        println!("Sending counter: {}", counter);

        rec.set_time_sequence("frame_tick", counter as i64);

        let points = grid(glam::Vec3::splat(-10.0), glam::Vec3::splat(10.0), 10);
        let colors = grid(glam::Vec3::ZERO, glam::Vec3::splat(255.0), 10)
            .map(|v| rerun::Color::from_rgb(v.x as u8, v.y as u8, v.z as u8));

        rec.log(
            "my_points",
            &rerun::Points3D::new(points)
                .with_colors(colors)
                .with_radii([0.5]),
        )?;

        std::thread::sleep(std::time::Duration::from_secs(1));
        counter += 1;

        if counter > 100{
            break;
        }
    }

    Ok(())
}
```

**Expected behavior**

I expected no errors in the viewer and that something is displayed

**Screenshots**

**Backtrace**

There is no crash in the demo code - the errors are in the viewer side, it logs the following every time it receives something

```
[2025-06-30T16:49:46Z ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
[2025-06-30T16:49:46Z ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
[2025-06-30T16:49:47Z ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
[2025-06-30T16:49:47Z ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
[2025-06-30T16:49:48Z ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column
[2025-06-30T16:49:48Z ERROR re_grpc_server] dropping LogMsg due to failed decode: Failed to read chunk: Bad chunk schema: Missing row_id column

```

**Desktop (please complete the following information):**
 - OS: Linux user 6.8.0-60-generic #63~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 22 19:00:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux 

**Rerun version**

```
rerun-cli 0.23.4 (default map_view nasm native_viewer release web_viewer) [rustc 1.84.0 (9fc6b4312 2025-01-07), LLVM 19.1.5] x86_64-unknown-linux-gnu 10350/merge b411df2, built 2025-06-26T13:13:49Z
Video features: av1 default ffmpeg nasm serde
```

**Additional context**

fortminors, 2025-06-30:
@Wumpf Opened the issue here as discussed in https://github.com/rerun-io/rerun/issues/2353

Wumpf, 2025-07-01:
I was fairly sure we do already and report that as a warning, it's probably just drowned out by the other error messages

closed 2025-07-02 by Wumpf