rerun-io/rerun #10459: [Tracking issue] Zombie todos

grtlr, 2025-07-01:
@Wumpf and I ran the zombie issue script and here are the results.

Please note that the issue author might not be the person that fixed the TODO.

## Zombie issues

* [ ] `// TODO(#7484): show something more useful here`
   * #7484 (Issue author: @Wumpf)
   * [`crates/viewer/re_data_ui/src/blob.rs#L231`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_data_ui/src/blob.rs#L232)
* [ ] `// TODO(#10226): improve fix the UI for built-in properties (`RecordingInfo`) vs custom ones.`
   * #10226 (Issue author: @grtlr)
   * [`crates/viewer/re_selection_panel/src/selection_panel.rs#L511`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_selection_panel/src/selection_panel.rs#L512)
* [ ] `// TODO(#4491): the projection breadcrumbs are wrong for nuscenes (but correct for arkit!),`
   * #4491 (Issue author: @abey79)
   * [`crates/viewer/re_selection_panel/src/item_heading_with_breadcrumbs.rs#L148`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_selection_panel/src/item_heading_with_breadcrumbs.rs#L149)
* [ ] `// TODO(#9134): Unify with `re_data_source::DataSource`.`
   * #9134 (Issue author: @grtlr)
   * [`crates/viewer/re_viewer/src/web_tools.rs#L82`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_viewer/src/web_tools.rs#L83)
* [ ] `// TODO(#4304): replace with a panel showing recent log messages`
   * #4304 (Issue author: @emilk)
   * [`crates/viewer/re_viewer/src/ui/top_panel.rs#L154`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_viewer/src/ui/top_panel.rs#L155)
* [ ] `//TODO(#7498): might be unneeded after the dataframe view update`
   * #7498 (Issue author: @abey79)
   * [`crates/viewer/re_component_ui/src/timeline.rs#L5`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_component_ui/src/timeline.rs#L6)
* [ ] `// TODO(#9903): Chunk descriptor referencing should be made trivial so that it doesn't matter whether we borrow or copy it.`
   * #9903 (Issue author: @Wumpf)
   * [`crates/viewer/re_view/src/chunks_with_descriptor.rs#L36`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_view/src/chunks_with_descriptor.rs#L37)
* [ ] `// TODO(#7608): All image data ingestion conversions should all be handled by re_renderer!`
   * #7608 (Issue author: @Wumpf)
   * [`crates/viewer/re_viewer_context/src/gpu_bridge/image_to_gpu.rs#L248`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_viewer_context/src/gpu_bridge/image_to_gpu.rs#L249)
* [ ] `// This is especially important if there's no miters on a line-strip (TODO(#829)),`
   * #829 (Issue author: @Wumpf)
   * [`crates/viewer/re_renderer/shader/lines.wgsl#L263`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_renderer/shader/lines.wgsl#L264)
* [ ] `// (potentially related to:) TODO(#7327, #7595): We don't necessarily have to enqueue full GOPs always.`
   * #7327 (Issue author: @jprochazk)
   * [`crates/viewer/re_renderer/src/video/player.rs#L309`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_renderer/src/video/player.rs#L310)
* [ ] `// TODO(#7608): to be replaced by re_renderer based on-input conversion.`
   * #7608 (Issue author: @Wumpf)
   * [`crates/viewer/re_renderer/src/renderer/rectangles.rs#L49`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_renderer/src/renderer/rectangles.rs#L50)
* [ ] `// TODO(#7608): Add rgb (3 channels!) formats.`
   * #7608 (Issue author: @Wumpf)
   * [`crates/viewer/re_renderer/src/resource_managers/image_data_to_texture.rs#L33`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_renderer/src/resource_managers/image_data_to_texture.rs#L34)
* [ ] `// TODO(#9917): Note that the view coordinates component may be logged by different archetypes which is why we do a name query here.`
   * #9917 (Issue author: @Wumpf)
   * [`crates/viewer/re_view_spatial/src/view_3d.rs#L288`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_view_spatial/src/view_3d.rs#L289)
* [ ] `// TODO(#9917): This is confusing and should be cleaned up.`
   * #9917 (Issue author: @Wumpf)
   * [`crates/viewer/re_view_spatial/src/transform_cache.rs#L1150`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_view_spatial/src/transform_cache.rs#L1151)
* [ ] `// TODO(#9917): This is confusing and should be cleaned up.`
   * #9917 (Issue author: @Wumpf)
   * [`crates/viewer/re_view_spatial/src/transform_cache.rs#L1176`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_view_spatial/src/transform_cache.rs#L1177)
* [ ] `// TODO(#7245): Use this instead of the above`
   * #7245 (Issue author: @Wumpf)
   * [`crates/viewer/re_view_spatial/src/transform_cache.rs#L1821`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_view_spatial/src/transform_cache.rs#L1822)
* [ ] `// TODO(#7245): Use this instead`
   * #7245 (Issue author: @Wumpf)
   * [`crates/viewer/re_view_spatial/src/transform_cache.rs#L2075`](https://github.com/rerun-io/rerun/blob/main/crates/viewer/re_view_spatial/src/transform_cache.rs#L2076)
* [ ] `// TODO(#9903): A reference to component descriptor should be enough since the returned iterator doesn't depend on it being alive.`
   * #9903 (Issue author: @Wumpf)
   * [`crates/store/re_chunk/src/iter.rs#L328`](https://github.com/rerun-io/rerun/blob/main/crates/store/re_chunk/src/iter.rs#L329)
* [ ] `# TODO(#2623): There should be a separate overridable `coerce_to_array` method that can be overridden.`
   * #2623 (Issue author: @abey79)
   * [`rerun_py/rerun_sdk/rerun/datatypes/time_range_boundary.py#L88`](https://github.com/rerun-io/rerun/blob/main/rerun_py/rerun_sdk/rerun/datatypes/time_range_boundary.py#L89)
* [ ] `// TODO(#9084): do we need this check? If so, how can we accomplish it?`
   * #9084 (Issue author: @emilk)
   * [`rerun_py/src/dataframe.rs#L892`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/dataframe.rs#L893)
* [ ] `// TODO(#9084): do we need this check? If so, how can we accomplish it?`
   * #9084 (Issue author: @emilk)
   * [`rerun_py/src/dataframe.rs#L963`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/dataframe.rs#L964)
* [ ] `// TODO(#9084): do we need this?`
   * #9084 (Issue author: @emilk)
   * [`rerun_py/src/dataframe.rs#L1021`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/dataframe.rs#L1022)
* [ ] `// TODO(#9084): do we need this check? If so, how can we accomplish it?`
   * #9084 (Issue author: @emilk)
   * [`rerun_py/src/catalog/dataframe_query.rs#L155`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/catalog/dataframe_query.rs#L156)
* [ ] `// TODO(#9084): do we need this check? If so, how can we accomplish it?`
   * #9084 (Issue author: @emilk)
   * [`rerun_py/src/catalog/dataframe_query.rs#L222`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/catalog/dataframe_query.rs#L223)
* [ ] `// TODO(#9084): do we need this?`
   * #9084 (Issue author: @emilk)
   * [`rerun_py/src/catalog/dataframe_query.rs#L270`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/catalog/dataframe_query.rs#L271)
* [ ] `//TODO(#9360): `dataset_from_url()``
   * #9360 (Issue author: @abey79)
   * [`rerun_py/src/catalog/catalog_client.rs#L224`](https://github.com/rerun-io/rerun/blob/main/rerun_py/src/catalog/catalog_client.rs#L225)
* [ ] `// TODO(#6832): Right now everything is crammed into a single struct array,`
   * #6832 (Issue author: @Wumpf)
   * [`rerun_cpp/src/rerun/archetypes/tensor_ext.cpp#L61`](https://github.com/rerun-io/rerun/blob/main/rerun_cpp/src/rerun/archetypes/tensor_ext.cpp#L62)
* [ ] `# * does memory usage induced by mp4 parsing stay low over time and doesn't accumluate (TODO(#7481): it should be smaller to begin with)`
   * #7481 (Issue author: @Wumpf)
   * [`tests/python/chunked_video/chunked_video.py#L12`](https://github.com/rerun-io/rerun/blob/main/tests/python/chunked_video/chunked_video.py#L13)
* [ ] `TODO(#7532): Some stuttering that can't be mitigated this way has been observed with H.264 video on Linux Firefox v130.0.`
   * #7532 (Issue author: @jprochazk)
   * [`docs/content/getting-started/troubleshooting.md#L135`](https://github.com/rerun-io/rerun/blob/main/docs/content/getting-started/troubleshooting.md#L136)
* [ ] ``
   * #7499 (Issue author: @nikolausWest)
   * [`docs/content/getting-started/data-out/export-dataframe.md#L50`](https://github.com/rerun-io/rerun/blob/main/docs/content/getting-started/data-out/export-dataframe.md#L51)
* [ ] ``
   * #7499 (Issue author: @nikolausWest)
   * [`docs/content/getting-started/data-out/explore-as-dataframe.md#L56`](https://github.com/rerun-io/rerun/blob/main/docs/content/getting-started/data-out/explore-as-dataframe.md#L57)
* [ ] `# TODO(#6670): Can't specify rr.components.FillMode.MajorWireframe right now, need to use batch type instead.`
   * #6670 (Issue author: @Wumpf)
   * [`examples/python/nuscenes_dataset/README.md#L144`](https://github.com/rerun-io/rerun/blob/main/examples/python/nuscenes_dataset/README.md#L145)

abey79, 2025-07-02:
> * `# TODO(#2623): There should be a separate overridable `coerce_to_array` method that can be overridden.`
>   * #2623 (Issue author: @abey79)
>   * [...]

@Wumpf I dug into this one and it turns out it's codegen'd. We have an entire branch for that one Union type 😅 Maybe there is a better way to go about it now?

abey79, 2025-07-03:
I dont understand that todo either tbh. I just noted that it was codegen'd (and as such is visible in the Rust codegen code). Also, this is about the one arrow union type we seem to have? So I was wondering if there was a way to nuke a whole lot more code than just this todo.

PR #10650 (2025-07-15): Fix TODO notes that refer to closed issues
### Related

* Fixes #10459

### What

A few minor doc fixes and refators along the way. Nothing big.
closed 2025-07-15 by Wumpf