rerun-io/rerun #10609: Default to WebGPU on Safari once stable

Wumpf, 2025-07-12:
Safari 18.5 (and lower) has WebGPU as an opt-in feature. Due to a (meanwhile fixed) bug in WebKit we're always defaulting to GL device on Safari (but you can still force it) even when Safari's experimental WebGPU support is enabled.

The bug has been fixed meanwhile, see 
* https://github.com/rerun-io/rerun/issues/8559

I have confirmed that Rerun works fine on WebGPU in the Safari Technology Preview 223 (WebKit 20622.1.18.11.5).

We should monitor further development and enable WebGPU in new enough Safari versions once we confirmed that a stable build of Safari works with Rerun @ WebGPU.

abey79, 2025-09-22:
As of Safari 26 running on macOS 15.6.1, I'm getting a blank screen and the following error.

[image]

abey79, 2025-09-22:
Correct. #11292 works if you didn't opt-in. But if you did, the experience is absolutely awful: black screen and no actionable error. Also, no way out but updating the url manually to force webgl, or disable opt-in.

Weirdly, the experience is actually _worse_ than it used to be (see https://github.com/rerun-io/rerun/issues/8559, there were glitches, but the viewer would otherwise run correctly).

Wumpf, 2025-09-25:
Bug for Safari 26 @ MacOS 15 regressing
https://bugs.webkit.org/show_bug.cgi?id=299510