rerun-io/rerun #10658: Air Traffic Data example is failing on local

pweids, 2025-07-16:
Could be something with my setup, not sure (M4 MBP). Here's the reproduction steps:
```
git checkout main
pixi run py-build-examples
pixi run -e examples air_traffic_data
```

And the error:
```
❯ pixi run -e examples air_traffic_data
DEV ENVIRONMENT DETECTED! Re-importing rerun from: /Users/paul/Development/rerun/rerun_py/rerun_sdk
[2025-07-16T00:06:22Z DEBUG rerun::commands::entrypoint] Detected 14 cores. Using 12 compute threads.
[2025-07-16T00:06:22Z DEBUG rerun::commands::entrypoint] Parsing memory limit for Viewer
[2025-07-16T00:06:22Z DEBUG re_memory::memory_limit] Setting memory limit to 36.0 GiB, which is 75% of total available memory (48.0 GiB).
[2025-07-16T00:06:22Z DEBUG rerun::commands::entrypoint] Parsing memory limit for gRPC server
[2025-07-16T00:06:22Z DEBUG rerun::commands::entrypoint] Server memory limit: 0B
[2025-07-16T00:06:22Z INFO  re_grpc_server] Listening for gRPC connections on 0.0.0.0:9876. Connect by running `rerun --connect rerun+http://127.0.0.1:9876/proxy`
[2025-07-16T00:06:22Z DEBUG eframe] Using the wgpu renderer
[2025-07-16T00:06:22Z DEBUG eframe::native::wgpu_integration] Event::Resumed
[2025-07-16T00:06:22Z DEBUG eframe::native::file_storage] Loading app state from "/Users/paul/Library/Application Support/rerun/app.ron"…
[2025-07-16T00:06:22Z DEBUG egui_wgpu::setup] Creating wgpu instance with backends Backends(VULKAN | GL | METAL)
[2025-07-16T00:06:22Z DEBUG egui_wgpu] Picked the only available wgpu adapter: backend: Metal, device_type: IntegratedGpu, name: "Apple M4 Pro"
[2025-07-16T00:06:22Z DEBUG re_renderer::context] wgpu adapter backend: Metal, device_type: IntegratedGpu, name: "Apple M4 Pro"
[2025-07-16T00:06:22Z DEBUG re_ui::hot_reload_design_tokens::design_token_access] Hot-reloading of design tokens enabled.
[2025-07-16T00:06:22Z DEBUG re_ui::hot_reload_design_tokens::design_token_access] Watching for file changes in /Users/paul/Development/rerun/crates/viewer/re_ui/data
[2025-07-16T00:06:22Z DEBUG re_viewer::app] Adding new log receiver: MessageProxy(ProxyUri { origin: Origin { scheme: RerunHttp, host: Ipv4(0.0.0.0), port: 9876 } })
[2025-07-16T00:06:22Z DEBUG tonic_web::service] kind="other h2" content_type=Some("application/grpc")
[2025-07-16T00:06:22Z DEBUG re_viewer::app] Trying to load blueprint for rerun_example_air_traffic_data from "/Users/paul/Library/Application Support/rerun/blueprints/rerun_example_air_traffic_data.rbl"
[2025-07-16T00:06:22Z DEBUG re_viewer_context::store_hub] Activating new blueprint fe304caad03f4c7ca8dad51e082ab035 for rerun_example_air_traffic_data; loaded from disk
Loading measurements:   0%|                                                                                                                                                               | 0/148 [00:00<?, ?it/s][2025-07-16T00:06:22Z DEBUG egui::context] Ignoring call request_discard, because max_passes=2. Requested from [crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, /Users/paul/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/egui-0.32.0/src/grid.rs:452 new Grid, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics, crates/viewer/re_ui/src/list_item/scope.rs:79 Missing re_ui::LayoutStatistics]
[2025-07-16T00:06:22Z DEBUG reqwest::connect] starting new connection: https://tile.openstreetmap.org/
Loading measurements: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 148/148 [00:06<00:00, 24.24it/s]
Computing area of interest...
Logging measurements:   1%|▉                                                                                                                                           | 8191/1205225 [00:00<00:11, 105694.41it/s]
Traceback (most recent call last):
  File "/Users/paul/Development/rerun/.pixi/envs/examples/bin/air_traffic_data", line 10, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/paul/Development/rerun/examples/python/air_traffic_data/air_traffic_data.py", line 465, in main
    log_everything(paths, args.raw, args.batch, args.batch_size)
  File "/Users/paul/Development/rerun/examples/python/air_traffic_data/air_traffic_data.py", line 407, in log_everything
    logger.process_measurement(measurement)
  File "/Users/paul/Development/rerun/examples/python/air_traffic_data/air_traffic_data.py", line 298, in process_measurement
    self.flush()
  File "/Users/paul/Development/rerun/examples/python/air_traffic_data/air_traffic_data.py", line 309, in flush
    self.log_position_and_altitude(group, icao_id)
  File "/Users/paul/Development/rerun/examples/python/air_traffic_data/air_traffic_data.py", line 347, in log_position_and_altitude
    *raw_coordinates.columns(),
     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/paul/Development/rerun/rerun_py/rerun_sdk/rerun/any_value.py", line 364, in columns
    inst = cls(drop_untyped_nones, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/paul/Development/rerun/rerun_py/rerun_sdk/rerun/any_value.py", line 291, in __init__
    raise ValueError(
ValueError: AnyValues should be initialized with at least one component. Please provide at least one keyword argument.
```

Given the error, I'm concerned a breaking change happened in `rr.AnyValues`, so raising this as a bug

grtlr, 2025-07-16:
@Wumpf and I ran into this behavior in CI before and we thought it was related to corrupted / unfinished downloads (some of the resources are a bit flaky). Our reasoning was that the implicit Numpy conversions fail if the data is incomplete.

If you have time, could you please delete the temporary assets and try again.

ntjohnson1, 2025-07-16:
FYI This looks like a bug in the example (or maybe it worked before and our api changed). [This line](https://github.com/rerun-io/rerun/blob/58a1897e69ab9e21c05fa8a44bc14e6910cace48/examples/python/air_traffic_data/air_traffic_data.py#L347) doesn't do what it appears we expect it to do. It looks like we expect it to return a column oriented view of the data [we gathered earlier](https://github.com/rerun-io/rerun/blob/58a1897e69ab9e21c05fa8a44bc14e6910cace48/examples/python/air_traffic_data/air_traffic_data.py#L335-L339) where instead its just an alternate constructor.

Wumpf, 2025-07-16:
this might have been used to establish an indicator previously

PR #10666 (2025-07-16): AnyValues Edge Cases and Air Traffic Example Fix
### Related
Closes #10658 
Somewhat a follow up to #10647 

### What
* I added a check for improper AnyValues usage (providing a value rather than kwarg) and that blew up the example since it was setup wrong
* Debugging this issue I hit
  * Numpy arrays of non-numeric types don't work for dlpack (check for BufferError)
  * Empty AnyValues are valid when used later, so I narrowed my check
  * Add an additional check for empty batches since that combined with my narrow check should prevent foot guns
  * Update our unit tests to better cover this