rerun-io/rerun #10708: failure to run `rr.notebook_show ()` in Google Colab notebook

glk0, 2025-07-19:
**Describe the bug**
Calling `rr.notebook_show ()` yields no output. JS errors reported in console.

**To Reproduce**
Run the following in a Colab notebook, on a CPU instance:
```python
!uv pip install rerun-sdk[notebook]

import rerun as rr
from numpy.random import default_rng

rr.init("example_dataviz")
rng = default_rng(12345)

positions = rng.uniform(-5, 5, size=[10, 3])
colors = rng.uniform(0, 255, size=[10, 3])
radii = rng.uniform(0, 1, size=[10])

points = rr.Points3D (positions, colors = colors, radii = radii)
rr.log ("random_pts", points)

rr.notebook_show(width=400, height=400) # !
```

**Expected behavior**
A visualization. If it's a bug and not something on my side, it must have been introduced recently because the code worked fine a couple days ago.

**Screenshots**
N/A

**Backtrace**
JS error messages in console:
```
index.js:777 Uncaught (in promise) Error: [anywidget] Failed to initialize model.
    at AbortSignal.eval (index.js:777:24)
widget.js:4442 using deprecated parameters for the initialization function; pass a single object instead
__wbg_init @ widget.js:4442
widget.js:748 Uncaught (in promise) LinkError: WebAssembly.instantiate(): Import #887 "wbg" "__wbindgen_closure_wrapper1338": function import requires a callable
    at __wbg_load (widget.js:748:42)
    at __wbg_init (widget.js:4450:49)
    at async load (widget.js:4498:3)
    at async WebViewer.start (widget.js:4558:27)
```

**Desktop (please complete the following information):**
- OS: `Linux 6.15.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 10 Jul 2025 17:10:18 +0000 x86_64 GNU/Linux`
- Browser: Chromium v38.0.7204.157-1

**Rerun version**
```
rerun-cli 0.24.0 (default map_view nasm native_viewer release web_viewer) [rustc 1.85.0 (4d91de4e4 2025-02-17), LLVM 19.1.7] x86_64-unknown-linux-gnu release-0.24.0 cf9dc2b, built 2025-07-17T14:19:01Z
Video features: av1 default ffmpeg nasm serde
```

**Additional context**
None

ntjohnson1, 2025-07-19:
I can confirm I see this locally. Building the development version locally off of main the problem goes away. Definitely looks like something in the 0.24.0 release. Currently looking at if it was an artifact of the release itself (building the wheels on main to see if things are still resolved) or if we happened to fix it on main and just need to cherry pick it over.

ntjohnson1, 2025-07-19:
Ok this looks to be a bug in our packaging. Either our source build environment pins something in a way we don't specify for the wheel build or our build script has issues.

Works on main as
```console
# clone repo
pixi install
pixi run py-build
pixi run py-build-notebook
# uncomment the line in the pixi.toml to install the notebook dependency
# launch a notebook and open a viewer
```
[Line to uncomment](https://github.com/rerun-io/rerun/blob/a1bce5156d46c9d4c9e5610c46bcae47fd09acf4/pixi.toml#L695)

Wheels don't work on main like
```console
uv venv --python=3.12
source .venv/bin/activate
pixi run py-build-wheel # for sdk wheels
uv pip install hatch
cd rerun_notebook
hatch build -t wheel .
uv pip install target/wheels/rerun_sdk-<>.whl rerun_notebook/rerun_notebook-<>/whl
uv pip install ipykernel # or might need jupyterlab depending how you open a notebook
# launch notebook and open viewer
```

ntjohnson1, 2025-07-19:
Ooh that's a great find! I actually was able to resolve locally by invalidating a build cache (but that wasn't very satisfying) but this makes me think we have a similar issue.

https://github.com/rerun-io/rerun/tree/nick/notebook_build

glk0, 2025-07-19:
> Hmm things seem to work with our last pre-release `pip install rerun-sdk==0.24.0a9 rerun-notebook==0.24.0a9`
I confirm on my side.

ntjohnson1, 2025-07-19:
Ok the viewer in the notebook doesn't actually return an error is just doesn't load so I can't just run auto bisect.

I put together a little script that I can run on a manual bisect that persists isolated environments in case I typo something
```bash
#!/bin/bash
deactivate
HASH=$(git rev-parse --short HEAD)
uv venv $HASH --python python3.12
source $HASH/bin/activate
uv pip install hatch ipykernel
rm target/wheels/rerun_sdk*.whl
rm rerun_notebook/rerun_notebook-*.whl
pixi run py-build-wheel
pixi run js-build-base
cd rerun_notebook
hatch build -t wheel .
uv pip install rerun_notebook-*.whl
cd ..
uv pip install target/wheels/rerun_sdk*.whl
```
The wheel build makes things much slower and I'm not totally certain its needed (checking for the error is a little tricky because it seems like once it works something caches and it keeps working so I figure better to do too much every step than to have to repeat). Also need to export `RERUN_NOTEBOOK_ASSET=serve-local` for the testing to make sure the right stuff gets loaded for non-released arbitrary commits.

Using that I see that even for the alpha9 commit I listed above with this recipe things fail. Somewhat surprisingly [this](https://github.com/rerun-io/rerun/pull/10111) passes. On inspection I kind of expected that to be the breaking change. I'll do a wider manual bisect later to see if that highlights our issue.

jleibs, 2025-07-20:
I started by taking a working copy of the wasm and just replacing the version hosted for 0.24.0, but that ended up breaking the main-site. Unfortunately, this ended up breaking the main-sight in a different wasm.

It looks like we expect the `.wasm` artifacts to be set up for `rerun.io/viewer` vs the notebook in different ways?

To keep everything working, I instead created a new asset: `https://app.rerun.io/version/0.24.0/re_viewer_bg_notebook.wasm` and then patched the hosted version of `widget.js` to point to that version instead. 

This is just a workaround to keep the lights on while we investigate the root cause of the divergence.

Of likely relevance, the 2 .wasms are vastly different sizes:
```
➜  gcloud storage ls -l gs://rerun-web-viewer/version/0.24.0/*.wasm
   9989892  2025-07-20T13:35:56Z  gs://rerun-web-viewer/version/0.24.0/re_viewer_bg.wasm
  36814914  2025-07-20T14:05:46Z  gs://rerun-web-viewer/version/0.24.0/re_viewer_bg_notebook.wasm
```

jleibs, 2025-07-20:
Also, the reason this works with alpha.9 is related to a path-sanitization inconsistency:
- The viewer retrieves: https://app.rerun.io/version/0.24.0-alpha.9/re_viewer_bg.wasm (10MB)
- The notebook retrieves: https://app.rerun.io/version/0.24.0a9/re_viewer_bg.wasm (35MB)

ntjohnson1, 2025-07-20:
I can confirm my clean uv env and on collab things work now. I'll mark this closed and open a new issue which should be more focused on what's happening in our build.

closed 2025-07-20 by ntjohnson1