rerun-io/rerun #10719: Rendering `Points3D` in spatial views with extreme aspect-ratios looks off

grtlr, 2025-07-22:
This was reported by Gautham on Discord: https://discord.com/channels/1062300748202921994/1397008726090580112/1397008726090580112

**Describe the bug**

Rendering `Points3D` in `SpatialView3D` with extreme aspect ratios (very wide view) seems off.

**To Reproduce**

This bug happens in `v0.24` and `v0.23.4` and can be reproduced in the web viewer.

Steps to reproduce the behavior:
1. Go to `nuscene` example
2. Configure 3D view to be very wide with little height
3. Points get stretched, in both world _and_ ui radii.

**Expected behavior**
At least for UI radii, the points should remain their 1:1 aspect ration and not be stretched.

**Screenshots**

UI points:
[image]

Scene points:
[image]

kpreid, 2025-07-22:
> Even when using scene points, is there a reason why the points look stretched when the viewer is stretched horizontally vs vertically?

There appears to be a bug with point size even in scene points, but even with that fixed, objects will *always* look stretched at the edge of a perspective projection.

* A perspective projection always has to choose a vertical and horizontal field of view (what angle is visible), and in order to avoid distorting shapes at the center, the relationship between them must match the aspect ratio (width : height ratio) of the viewport. Typically, this is handled by defining a vertical FOV angle, and then letting the horizontal FOV be whatever keeps the center undistorted (i.e. circles are not ellipses, when they appear at the center). Rerun appears to be doing this.
* Large FOV always causes objects closer to the edges of the image to be more stretched in that axis. This is not a bug; it is fundamentally how the math of perspective works out. Theoretically, to have an undistorted view, FOV should always be set to match the angular size of the image on your monitor as you are looking at it with your actual eyes, but almost nobody does this because it makes it hard to see very much.
* So, when you have a tall viewport, it has a very small horizontal FOV with low stretch (so low you don't notice), and when you have a wide viewport, it has a very large horizontal FOV with a large stretch.

Rerun could “fix” the excessive distortion you see by, for example, deciding that if the 3D view's aspect ratio is wider than it is tall, then the vertical FOV should vary while the horizontal FOV is kept fixed. But, that's more like an improved heuristic than a bug-fix, and it will mean that moderately wide viewports show less content than they would otherwise.

kpreid, 2025-07-22:
Not yet — see #2497 and #5234.